schreibfaul1/ESP32-MiniWebRadio #270: platformio esp32-a1s upload

newbee90, 2023-08-04:
leider bekomme ich die Software mit PlatfornIO nicht auf das Board, sagt beides mal es ist der falsche Chip

Arne65, 2023-08-04:
Hast du die Widerstände alle entfernt laut der Anleitung ? Das sollte zwar unerheblich sein für die Programmierung abermals weiß ja nie . Du bist auch auf dem UART ?

newbee90, 2023-08-04:
ich Depp, ich muß vor dem einstecken die boot Taste drücken, sonst läßt sich das Teil nicht programmieren

schreibfaul1, 2023-08-04:
Guck mal, ob unten in der Statuszeile `Default (ESP32-MiniWebRadio)` steht. Falls ja, ändere das bitte in `env:esp32(ESP32-MiniWebRadio)`

newbee90, 2023-08-05:
ich habe alle Variationen durch, default, esp32, esp32s2, kein upload, boot gedrückt beim Einstecken, boot gedrückt gehalten und reset gedrückt
Serial port COM4
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b8:d6:1a:58:24:d0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00050000 to 0x00372fff...
Compressed 18960 bytes to 13072...

A fatal error occurred: Packet content transfer stopped (received 8 bytes)
*** [upload] Error 2
============ [FAILED] Took 50.69 seconds ====================
Environment    Status    Duration
-------------  --------  ------------
esp32          FAILED    00:00:50.693
================ 1 failed, 0 succeeded in 00:00:50.693

Arne65, 2023-08-05:
Da findest Suhler alles was brauchst

https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/additional_info/AI%20Thinker%20A1S%20Board.pdf

newbee90, 2023-08-05:
da finde ich wie das Display angeschlossen wird aber nicht welche Werte ich für dieses Raspberry XPT2046 brauche für die common.h

Arne65, 2023-08-05:
Common.h Definition steht doch auf Seite 2

unten auf der Seite.  Next

newbee90, 2023-08-05:
das meine ich, die Werte brauche ich

#define DECODER             4                     // (0)VS1053 , (1)MAX98357A PCM5102A... (2)AC101 (3)ES8388 (4)WM8978
#define TFT_CONTROLLER      4             // (0)ILI9341, (1)HX8347D, (2)ILI9486a, (3)ILI9486b, (4)ILI9488, (5)ST779 (6)ST7796RPI
#define DISPLAY_INVERSION   0             // (0) off (1) on
#define TFT_ROTATION        1                // 1 or 3 (landscape)
#define TFT_FREQUENCY       40000000                        // 27000000, 40000000, 80000000
#define TP_VERSION          4                // (0)ILI9341, (1)ILI9341RPI, (2)HX8347D, (3)ILI9486, (4)ILI9488, (5)ST7796, (3)ST7796RPI
#define TP_ROTATION         1                               // 1 or 3 (landscape)
#define AUDIOTASK_CORE      1                               // 0 or 1
#define AUDIOTASK_PRIO      2                               // 0 ... 24  Priority of the Task (0...configMAX_PRIORITIES -1)
#define SDMMC_FREQUENCY     20000000              // 40000000, 2000000, 10000000, not every SD Card will run at 40MHz

Arne65, 2023-08-05:
#define DECODER 4 hängt davon ab ab welcher Chip drauf  ist   bei mir ES8383
 
#define TFT_CONTROLLER                                                                           ILI9486a

#define  TP_VERSION                                                                                    ILI 9486

Alle anderen Werte kannst du erstmal so lassen 

das Display hat mir nerven gekostet um es zum laufen zu bringen

newbee90, 2023-08-05:
Decoder habe ich alle Werte von 0 bis 4 durch, kein Ton auf der Kopfhörerbuches, Weboberfläche ist da und Musiktitel wird auch angezeigt auch im Terminal

Arne65, 2023-08-05:
Hast du in Common.h alles eingetragen alles auch -1 und pin 25,26 getauscht je nach verwendetem DAC ?

newbee90, 2023-08-05:
Danke schon mal für deine Hilfe
Ton ist jetzt da nachdem ich ALLE Zahlen, so wie sie auf der Liste stehen, geändert habe, 
nur wenn ich das Display anstecke geht nichts mehr und im Terminal kommt, 11562][E][Wire.cpp:526] write(): NULL TX buffer pointer

newbee90, 2023-08-05:
Angeschlossen habe ich es so
ESP   Display  
22      24
5        18
12      11
13      26
23      19
19      21
18      23

schreibfaul1, 2023-08-05:
Was hast du genau für ein Display, für Rasperrys gibt es verschiedene.
Eigenartig ist die Meldung `11562][E][Wire.cpp:526] write():` die auf I2C hindeutet. I2C wird nur intern zwischen dem ESP und den Audiochip benötigt und ist nicht nach außen geführt.
Verlinke bitte hier ein Bild des Displays.

newbee90, 2023-08-05:
hier ist das Bild
https://ibb.co/K2TbXMM

schreibfaul1, 2023-08-05:
Damit gibt es keine Probleme, hab selber dieses Diaplay won Waveshare. 
[image]

newbee90, 2023-08-05:
oben habe ich ja geschrieben wie ich es angeschlossen habe, es sind ja nur 7 Kabel plus Betriebsspannung, das habe ich schon 5x überprüft und Kurzschlüsse habe ich auch keine drauf, das Bild hast du mir schon geschicht weil ich danach gefragt hatte.

newbee90, 2023-08-05:
jetzt habe ich im Terminal
         Connected!
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

Arne65, 2023-08-05:
Ich habe denke ich das gleiche Display wie  Du

[image]

Das Einzige was ich geändert habe ist in 

Ich hatte immer einen weißes Display

Common.h   TP_IRQ =13    und TP_CS =12 und dann am Display getauscht dann hat es funktioniert ich nutze es aber derzeit nicht da mir Befestigungspunkte für das Display fehlen

Arne65, 2023-08-05:
Ich benutze ein Display mit dem ILI 9488 diese haben immer auf Anhieb funktioniert 

[image]

mit der obigen beschriebenen Änderung auch am a1s

newbee90, 2023-08-05:
bringt bei mir keinen Unterschied, das andere Display ist halt größer, das rote habe ich nur als 2,8"

Arne65, 2023-08-05:
ungültiger Header: 0xffffffff sieht für mich so aus, als ob der Flash gelöscht ist oder leer

newbee90, 2023-08-05:
die Meldung kommt aber nur wenn die Verbindung von 12 und 13 zum Display gesteckt ist

steve6375, 2023-08-05:
Did you see the comment in common.h about a 1k resistor for pin 12?

newbee90, 2023-08-05:
#define VS1053_SCK    -1  // VS1053     (HSPI) (sometimes we need a 1k resistor against ground)
I have no VS1053, its a ES8388

Arne65, 2023-08-05:
Ohne 12 und 13 funktioniert es dann soweit ?

Arne65, 2023-08-05:
S4 und S5 sind ON ?

newbee90, 2023-08-05:
1,2,3 off
4.5 on
macht keinen Unterschied,

Arne65, 2023-08-05:
Und die Widerstände sind raus aus dem a1s ?

newbee90, 2023-08-05:
die Widesstände sind raus
[image]

schreibfaul1, 2023-08-05:
Es sieht aus, als wenn GPIO0 beschaltet ist, aber Reset nicht,
RST wird mit dem Display verbunden und GPIO0 ist intern MCLK und bleibt frei

newbee90, 2023-08-05:
RST ist jetzt angeschlossen und es geht auch das Display, GIO0 ist nicht belegt,
wenn 12 angeschlossen ist geht gar nichts,

Arne65, 2023-08-05:
kannst mal folgendes versuchen 12 trennen Rest  drücken dann 12 wieder anschließen 

sollte es dann funktionieren 12 /13 tauschen und in Common.h

newbee90, 2023-08-05:
12 und 13 tauschen hat Arne65 schon geschrieben, hat nichts gebracht, sobald die 12 angeschlossen ist geht nichts mehr
egal ob 12 und 13 vertauscht sind oder nicht
wie müsser den S1 bis 3 stehen?

schreibfaul1, 2023-08-06:
GPIO2 und GPIO12 sind "strapping pins" die über einen internen Pulldown Widerstand des ESP beim Booten aul Low gezogen werden. Wenn beim Start GPIO12 (oder GPIO2) als High erkannt wird läuft der Bootvorgang nicht korrekt. Was ist, wenn du GPIO12 beim Start unbeschaltet lässt und danach erst verbindest (zeitlich kein Problem, ist nur der Eingang für TP_IRQ)? Falls es dann funktioniert, sind die vermutlich vorhandenen PullUps des Displays zu stark.

newbee90, 2023-08-06:
sobald das Startbild erscheint kann ich GPIO12 anschließem
ich habe 10k als Pullup am Display gemessen, 
ich warte auf die neuen Displays

schreibfaul1, 2023-08-06:
Wahrscheinlich kannst du den Pullup mit einem gleichgroßen Widerstand gegen Masse kompensieren.

steve6375, 2023-08-06:
IO12 has R37 pullup and is connected to SD card.
So maybe R37  is causing issues?

newbee90, 2023-08-06:
The problem is only when a display is connected, so I wait for different displays

steve6375, 2023-08-06:
The SD card has a pull up and the display has a pull up, so maybe that is much.
R37 could be removed to see what affect it has?

newbee90, 2023-08-06:
I see R37 in the datasheed but can´t find it on the board, I find R23, between 3,3V pin and GPIO12 its over 2,3M, so maybee R37 is missing

steevemadison, 2023-08-11:
here what i did to make it work , without problem , i put a wire at gpio4 ,at gpio36 and gpio12 is use for ir , then my common.h config is.

        #define TFT_CS        22
        #define TFT_DC         5
        #define TFT_BL         4
        #define TP_IRQ        36
        #define TP_CS         13
        
        #define SD_MMC_D0      2
        #define SD_MMC_CLK    14
        #define SD_MMC_CMD    15

        #define IR_PIN        12

        #define TFT_MOSI      23
        #define TFT_MISO      19
        #define TFT_SCK       18
    #if DECODER == 0
        #define VS1053_CS     -1
        #define VS1053_DCS    -1
        #define VS1053_DREQ   -1
        #define VS1053_MOSI   -1
        #define VS1053_MISO   -1
        #define VS1053_SCK    -1
    #else
        #define I2S_DOUT      26
        #define I2S_DIN       35
        #define I2S_BCLK      27
        #define I2S_LRC       25
        #define I2S_MCLK       0
    #endif
        #define I2C_DATA      33
        #define I2C_CLK       32
        #define SD_DETECT     34
        #define HP_DETECT     39
        #define AMP_ENABLED   21

newbee90, 2023-08-11:
Perfect, thats all I need

closed 2024-01-21 by github-actions[bot]