schreibfaul1/ESP32-MiniWebRadio #272: Search function in Web UI

steevemadison, 2023-08-05:
Hi , i've updated my webradio to version 2.8.3b and the search by tag , by country , give me an error

Access to XMLHttpRequest at 'https://de1.api.radio-browser.info/json/tags' from origin 'http://192.168.0.100' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource

with version 2.8.1e , there is no error , the search tab is working , anyone got a clue

schreibfaul1, 2023-08-05:
Something has changed in the radio browser or the original URL is no longer available, a different json URL is used.

steevemadison, 2023-08-05:
Hi , thank's for your quick update , everything back to normal , search tab is working like before

thumb's up .

closed 2023-10-06 by steevemadison