schreibfaul1/ESP32-MiniWebRadio #274: Esp32 S3 not available

GrazAnt, 2023-08-10:
I am unable to find a Esp32 S3 AI thinker.
Could an ESP32-S3-DevKitC-1-N16R8V be used as a replacement? 
As a reference:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html
And if it can be done, then, would it be possible to have the pin configuration or the schematic in order to build it correctly?

Thank you very much.

valera39, 2023-08-10:
Of course you can, I have one radio built on an ESP32 S3 N16R8. The connection is similar to Al Thinker. In the file common.h there is a pinout and in the archive with the project there is a picture with connection.

TubaFonse, 2023-08-11:
@GrazAnt 
I'm also using ESP32-S3-DevKitC-1-N16R8 (Chip: ESP32-S3-WROOM-1-N16R8). I use the I2S-DAC PCM5102A, but I had to change some GPIO's against the actual common.h (CONFIG_IDF_TARGET_ESP32S3)! 
You have to change some GPIO's when using WROOM-1-N16R8 Core:  
1.) Due to the internal connected Octal SPI PSRAM and Octal SPI Flash there are some GPIO's not usable (33-37), see datasheet. 
2.) Don't use GPIO 3 for DAC, since it is the "Strapping-Pin" to select USB/JTAG-Port (UART/USB); e.g. use GPIO 5 as I2S_BCLK. 
3.) Don't forget to configure platformio.ini correctly [env:esp32s3]: choose 
board_build.partitions = miniwebradio16MB.csv   ; 16MB Flash 
board = ESP32-S3-DevKitC-1-N16R8       ; 16MB Flash, 8MB octo PSRAM  *1)
 *1)   in menuconfig the type of PSRAM must be changed from QUAD to OCTAL!
;       component config ---> ESP32S3-Specific ---> SPI RAM config ---> MODE (QUAD/OCT)... ---> (x) Octal Mode PSRAM
good luck! 
@valera39 
you wrote: ... in the archive with the project there is a picture with connection
1.) are there the connection shown special for ESP32 S3 N16R8 (SOC system on chip)?
2.) WHERE can I find that archive / picture? Pls show me the link ... sorry I did'nt find that project archive.
3.) do you also have a pinout for configuration VS1053 and ESP32 S3 N16R8, which is working? 
You know, there are the restricted GPIO's 19, 20, (USB) 38 or 48 (RGB-LED) ...

GrazAnt, 2023-08-11:
@valera39 @TubaFonse
Thank you both very much, I will try out TubaFonse's advise in the next days and then let you know the results.
When I'm done I'll share my 3d design for the radio case with you all as thanks.

valera39, 2023-08-11:
@GrazAnt, here's my config for PCM 5102:
#ifdef CONFIG_IDF_TARGET_ESP32S3
    // Digital I/O used
        #define TFT_CS         8
        #define TFT_DC        12
        #define TFT_BL        10 // at -1 the brightness menu is not displayed
        #define TP_IRQ        39
        #define TP_CS         15
        #define SD_MMC_D0     11
        #define SD_MMC_CLK    13
        #define SD_MMC_CMD    14
        #define IR_PIN         4
        #define TFT_MOSI      18  // TFT and TP (FSPI)
        #define TFT_MISO       2  // TFT and TP (FSPI)
        #define TFT_SCK       17  // TFT and TP (FSPI)
    #if DECODER == 0
        #define VS1053_CS     48
        #define VS1053_DCS    16
        #define VS1053_DREQ   38
        #define VS1053_MOSI   20  // VS1053     (HSPI)
        #define VS1053_MISO   19  // VS1053     (HSPI)
        #define VS1053_SCK    21  // VS1053     (HSPI) (sometimes we need a 1k resistor against ground)
    #else
        #define I2S_DOUT       9
        #define I2S_DIN       -1
        #define I2S_BCLK       3
        #define I2S_LRC       46
        #define I2S_MCLK      -1
    #endif
        #define I2C_DATA      -1  // some DACs are controlled via I2C
        #define I2C_CLK       -1
        #define SD_DETECT     -1  // some pins on special boards: Lyra, Olimex, A1S ...
        #define HP_DETECT     -1
        #define AMP_ENABLED   -1
Everything works fine. The picture I used was for ESP32 Al Thinker, I didn't notice any difference https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/additional_info/ESP32-S3%20pins.jpg

GrazAnt, 2023-08-12:
@TubaFonse 

I'm sorry but I just can't find :
*1) in menuconfig the type of PSRAM must be changed from QUAD to OCTAL!
(where do i find this option ?)

and I get this error:
surely due to the fact that I have not configured the spiram

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420e4ba6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
E (554) psram: PSRAM ID read error: 0x00ffffff    <<<<<<< error
E (554) spiram: SPI RAM enabled but initialization failed. Bailing out.  <<<<<<<<<<< error
[   570][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled    OK
E (64) spiram: SPI RAM not initialized  <<<<<<<<<<<< error!!

abort() was called at PC 0x420617a4 on core 1

thank you very much and sorry for the inconvenience

GrazAnt, 2023-08-12:
@valera39 
many thanks to you too
now the problem is the configuration of the spiram
which I just don't know, I just started with platformio!!!

valera39, 2023-08-12:
[GrazAnt](https://github.com/GrazAnt)
Read it here #184, #180, Wolle explained in detail

GrazAnt, 2023-08-16:
Thank you all very much but I couldn't manage to compile the code, anyway I have published my project on the case on thingyverse at this address: 
https://www.thingiverse.com/thing:6172929
I also wanted to ask to @schreibfaul1 if I can post the github as a construction reference.
thanks

GrazAnt, 2023-10-05:
I'm still struggling, thank you anyway: I want to solve it myself.

GrazAnt, 2023-10-12:
I'm back just to say this:
what is written above about psram, the wires and all the advices are correct and they helped me resolve my problems and learn something new.
Now I have an "almost" perfect internet radio!!!
THANK YOU EVERYONE

closed 2023-10-12 by GrazAnt