schreibfaul1/ESP32-MiniWebRadio #278: Artifacts

valera39, 2023-08-15:
Hello Wolle!
On the new version, there are artifacts on the screen when switching clocks. More details in the video
https://github.com/schreibfaul1/ESP32-MiniWebRadio/assets/96370186/b2f1db11-387b-4099-95ea-215f4fe84064

schreibfaul1, 2023-08-15:
Thank you for pointing this out. I think this only happens when switching between RADIO and CLOCK with the IR. The new buttons are no longer bitmaps, but JPGs. This reduces the file size and thus ensures faster loading. However, JPGs must be divisible by 8 in width and height. The old buttons were 60x60px, now 56x56px. And that will explain the artefacts.

valera39, 2023-08-15:
Downloaded the updated version. Artifacts in the form of the rest of the clock segment are still present, both when controlling with the remote control and via touchpad
https://github.com/schreibfaul1/ESP32-MiniWebRadio/assets/96370186/cf708849-e6bc-42e0-8d42-283fd2139290

schreibfaul1, 2023-08-15:
This is strange. The VU meter is drawn only once at the beginning (or when switching to RADIO) and is only filled afterwards. Please check whether the file "level_bar.jpg" is on the SD card under common/m (or /s). It is also possible that the file is damaged.
[image]

valera39, 2023-08-15:
Turns out I had level_bar.bmp, I didn't replace it with level_bar.jpg. Everything is fine now. Thanks!
One more question. Were you able to create a list of stations on the screen? Do you plan to implement this?
Please reply.

closed 2023-08-19 by valera39