schreibfaul1/ESP32-MiniWebRadio #286: Web UI on Android Smartphones or Tablets: next/previous Station Button switches two stations forward or back

TubaFonse, 2023-08-24:
Dear Wolle -I need your help: 
This issue I only get when I use the Web UI on my Android Smartphones or Tablets to switch the radio station one forward or one back: 
e.g. 
1.) I press the arrow-right button one time to switch Station 5 forward to get the next Station (6), but it skips 6 and switch to Station 7.  
2.) I press the arrow-left button one time to switch Station 7 back to get the previous Station (6), but it skips 6 and switch to Station 5.  
I used Chrome and Firefox on Android, but I got always the same problem. 

No such problem on my Windows10 PC to switch the Stations in the Web UI (Chrome, Firefox, Edge)
[image]

schreibfaul1, 2023-08-25:
On systems with mouse operation there are the html events "onmousedown" and "onmouseup", on platforms like Android "ontouchstart" and "ontouching". Probably on newer devices/versions for touch operation the mouse events are also emulated. On my tablet I was able to reproduce the error. I have now used "onclick" for previous and next station.

TubaFonse, 2023-08-26:
Dear Wolle, the "next/previous station problem" is solved with current software version - many, many thanks!!!

closed 2023-08-26 by TubaFonse