schreibfaul1/ESP32-MiniWebRadio #287: Issue only with ESP32-S3-DevKitC-1-N8R2: no Radio Stations found

TubaFonse, 2023-08-24:
Dear Wolle, 
maybe you have an idea, why I only get this "Radio"-problem with ESP32-S3-DevKitC-1-N8R2 (WROOM-1) 
Flash 8 MB (Quad SPI) and PSRAM 2 MB (Quad SPI).:

23:31:31 WiFI_info:   RSSI is -52 dB
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[   788][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

ESP32 Chip: ESP32-S3
Arduino Version: 2.0.11
ESP-IDF Version: 4.4.4
MiniWebRadio     Version 2.10b Aug 23/2023
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 8388608 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 20 MHz
TFT speed 40 MHz
PSRAM total size: 2097152 bytes

00:00:00
00:00:00        ***************************
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 15173.0 MB by 15193.5 MB free
00:00:00 setup: ....  seek for stations.csv
00:00:00 setup: ....  stations.csv found
00:00:00 setup: ....  seek for WiFi networks
[  1445][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: FRITZ!Box7490LF
[  7514][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  7515][I][WiFiMulti.cpp:119] run(): [WIFI] 8 networks found
[  7516][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: C8:0E:14:0C:93:11 SSID: FRITZ!Box7490LF Channel: 11 (-48)
[  7557][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 202 - AUTH_FAIL
[  7635][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
00:00:06 WiFI_info:   Connecting WiFi...
00:00:06 setup: ....  connected to FRITZ!Box7490LF, IP address is 192.168.178.37
00:00:06 ftpServer:   Buffers allocated: 4622 bytes
00:00:06 rtime_info:  Initializing SNTP
00:00:06 rtime_info:  Waiting for system time to be set... (1/10)
23:31:46 rtime_info:  23:31:46
23:31:46 audiotask:   is pinned to core 1
23:31:46 audiotask:   priority is 2
23:31:46 setup: ....  Number of saved stations: 14
23:31:46 setup: ....  current station number: 7
23:31:46 setup: ....  current volume: 21
23:31:46 setup: ....  last connected host: http://192.168.178.24:50002/m/NDLNA/6221.mp3
23:31:46 setup: ....  connection timeout: 500 ms
23:31:46 setup: ....  connection timeout SSL: 2000 ms
23:31:46 action: ...  current volume is 21
[  9770][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_007 NOT_FOUND
23:31:46 action: ...  switch to station 7
23:31:46 AUDIO_info:  Connect to new host: "http://not_found"
23:31:46 AUDIO_info:  PSRAM found, inputBufferSize: 638965 bytes
23:31:46 AUDIO_info:  buffers freed, free Heap: 171067 bytes
[  9845][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for not_found
23:31:46 AUDIO_info:  Request http://not_found failed!
23:31:46 StreamTitle:
23:31:47 WiFI_info:   RSSI is -46 dB
23:31:47 DLNA_info:   SSDP search for media servers started
23:31:51 DLNA_info:   SSDP query discovered 2 media servers
23:31:51 DLNA_info:   connected successfully to 192.168.178.24:50001
23:31:51 DLNA_info:   connected successfully to 192.168.178.29:2869
23:31:51 ftpServer:   Ftp server waiting for connection on port 21
23:32:36 WiFI_info:   RSSI is -51 dB
23:32:37 WiFI_info:   RSSI is -47 dB
23:44:23 WiFI_info:   RSSI is -51 dB
[963208][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_008 NOT_FOUND
23:47:40 action: ...  switch to station 8
23:47:40 AUDIO_info:  Connect to new host: "http://not_found"
23:47:40 AUDIO_info:  buffers freed, free Heap: 168107 bytes
[963237][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for not_found
23:47:40 AUDIO_info:  Request http://not_found failed!
23:47:40 StreamTitle:
[971774][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_009 NOT_FOUND
23:47:48 action: ...  switch to station 9
23:47:48 AUDIO_info:  Connect to new host: "http://not_found"
23:47:48 AUDIO_info:  buffers freed, free Heap: 168107 bytes
[971799][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for not_found
23:47:48 AUDIO_info:  Request http://not_found failed!
23:47:48 StreamTitle:
[978883][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_010 NOT_FOUND
23:47:55 action: ...  switch to station 10
23:47:55 AUDIO_info:  Connect to new host: "http://not_found"
23:47:55 AUDIO_info:  buffers freed, free Heap: 168107 bytes
[978908][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for not_found
23:47:55 AUDIO_info:  Request http://not_found failed!
23:47:55 StreamTitle:

schreibfaul1, 2023-08-25:
Hello TubaFonse,
the problem will be in the NVS. The stations are saved in the NVS. Even if the programm is reloaded, the content of the NVS remains untouched. In Platfornio, please go to the ant on the left and then to Platform/Erase Flash. This will only work if the serial monitor is closed, if necessary, put the monitor in the dustbin first. This will delete all partitions. If you then upload the programme and start it, the empty NVS will be recognised and refilled with the content of stations.csv. I found another error, because this did not work in every case. Please use the current software version.

TubaFonse, 2023-08-26:
Hello Wolle, 
many thanks for your quick response and your great work - I used the current software version - and the problem is solved!

closed 2023-08-26 by TubaFonse