schreibfaul1/ESP32-MiniWebRadio #288: Audio Player stops Play after one Song

AVRsteffen, 2023-08-28:
Hello wool,
Please help me. I've been trying for 2 days now to have the audio player play all the songs one after the other up to the last song in a folder. I tried to modify the "void audio_eof _mp3()".
After the "eof" I can load the new audio file and show it on the display. But unfortunately I can't play the next song.
As soon as I try to run "SD_playFile(nextAudioFile.path())" the ESP32 hangs.

Please help me! What am I doing wrong?

Here my changes:

`void audio_eof_mp3(const char* info) { // end of mp3 file (filename)`
`   _f_eof = true;`
`    _f_isFSConnected = false;`
`    if(startsWith(info, "alarm")) _f_eof_alarm = true;`
`    SerialPrintfln("end of file: " ANSI_ESC_YELLOW "%s", info);`
`          if(_state == PLAYER || _state == PLAYERico) {`
`		clearLogo();`
`		clearStationName();`
`//		stopSong();`
`		nextAudioFile = getNextAudioFile();`
`               showFileName(nextAudioFile.name());`
`//		SD_playFile(nextAudioFile.path());`
`//		connecttoFS(nextAudioFile.path());`
`		SerialPrintfln("start next file: " ANSI_ESC_GREEN "%s", nextAudioFile.path());`
` `		
`		if(_f_isFSConnected && nextAudioFile){`
`			free(_lastconnectedfile);`
`			_lastconnectedfile = strdup(nextAudioFile.path());`
`			_resumeFilePos = 0;`
`		} else {`
`			SerialPrintfln("end of file Folder");`
`			webSrv.send("SD_playFile=end of audiofile");`			
`		}`
`    }`
`//    webSrv.send("SD_playFile=end of audiofile");`
`}`

Please help me! What am I doing wrong?

schreibfaul1, 2023-08-28:
Hello Steffen,
a simple possibility is to create an m3u playlist. There is an example on the SD card (extract from Content_on SD_card.zip if necessary). You can create an m3u file easily with a text editor, but it is more convenient with the VLC Player. Simply drag the desired files (locally, from the server in the home network or Internet) into the VLC Player and press Create Playlist. Then transfer them to the SD card with Filezilla.

AVRsteffen, 2023-08-28:
Yes, I know the way via a playlist.
But there must also be a way after an "eof" to load and play the next one - if available.
Or is that to difficult?

schreibfaul1, 2023-08-28:
I know what you mean, but it's not simple. The event `audio_eof_mp3(const char* info) ` is called by the audio task. You can't directly control the Adruino "loop" task with it. This will lead to collisions or a system crash. You need a message queue for the intertask communication. If it is only a question of status information, a flag is sufficient, and I have also used this. _f_eof is such a flag, which is periodically queried by the loop task.

AVRsteffen, 2023-08-29:
Okay, I figured it wouldn't be easy. But it should also work without a playlist.
Let's see if I got it right:
- The audio task takes care of playing the song completely independently.
- When done, a message is sent from the audio core to the Arduino core (eof) so that `void audio_eof_mp3()` is called and processed in the other core

Processing in separate cores doesn't make it any easier. Can't the Audio Core take care of everything? Also about reloading the next song? Or does it lose the "quit" or "cancel" control?

schreibfaul1, 2023-09-16:
There are different possibilities of what the "next" file should be. Do you mean the next audio file within a folder? I could add a function that plays all files within a folder without much effort.

valera39, 2023-09-16:
Seconded, that would be great.
Do you plan to develop your project further? Just haven't had any new ideas in a while

AVRsteffen, 2023-09-16:
Hello Wolle

Thanks for responds my question. Yes, like that. I guess it is perfect to have a function for play files within a folder. Additional a function to choose prefious/next file in folder. That will be perfect

steve6375, 2023-09-16:
Will it also loop and keep playing whole folder? In alphabetical order?
What about a random play forever function within the folder?
;-)

schreibfaul1, 2023-09-16:
I have first realised this for the website. The context menu for folders now also has the entry "Play All".
[image]
I used the function `openNextFile()`, unfortunately not alphabetically, the sequence of the files as they are stored on the SD card is used.

Important: With the new Arduino version from 2.0.12, the saved audio libraries no longer work. These must be reloaded after synchronisation with "Full clean".
[image]

valera39, 2023-09-18:
Dear Wolle, can I add a "play all" button to the display so I don't have to use the web page?

AVRsteffen, 2023-09-19:
My new branch dont work with "All Play" in Web Interface.. I dont know what I do wrong. I made a new 'Clone git project'  in a new project folder. In "index.js.h" are that changes in line 329-336 (Play All). But my Web Interface show me only "Play" and not "Play All"

Something I do wrong with platformio

schreibfaul1, 2023-09-19:
There must still be old data in the browser cache

AVRsteffen, 2023-09-19:
Can I clear with "Erase Flash"?

schreibfaul1, 2023-09-19:
I think it will be old data in the web browser that does not update automatically. In Chrome, this is called "Clear cached images and files".

AVRsteffen, 2023-09-19:
Okay, in Google Chrome at PC it works correct. But in Samsung Browser at phone isn't work..

Some ideas why?

schreibfaul1, 2023-09-23:
It is now possible to play audio files within a folder sequentially and in alphabetical order. There is a new button and a new logo for this. The graphics for this must be unpacked from the content_on_SD_card.zip and written to the SD card.
https://github.com/schreibfaul1/ESP32-MiniWebRadio/pull/299

closed 2025-02-16 by github-actions[bot]