schreibfaul1/ESP32-MiniWebRadio #289: mDNS name in local network

kpbicka, 2023-08-29:
Hello, Wolle!

It would be very convenient to open the radio page in the browser always with one link, for example, miniwebradio.local, because the device periodically receives a new address from DHCP. To avoid this problem, there is a multicast DNS (mDNS) protocol resolves hostnames to IP addresses within small networks that do not include a local name server. It is a zero-configuration service, using essentially the same programming interfaces, packet formats and operating semantics as unicast Domain Name System (DNS). some other projects on ESP microcontroller have mdns support. 

Please include this request in the list of future tasks. Thank you very much for the development of this project and its support.

kerat71, 2023-10-10:
Hi, how to use this? I write "miniwebradio.local" in Firefox address bar, but nothing is happen. Thanks!

kpbicka, 2023-10-10:
> Hi, how to use this? I write "miniwebradio.local" in Firefox address bar, but nothing is happen. Thanks!

http://miniwebradio.local works fine

[image]

closed 2025-02-16 by github-actions[bot]