schreibfaul1/ESP32-MiniWebRadio #292: Psram error with V2.11

ve2cfb, 2023-09-01:
Hello Wolle,
i was running V2.10C and everything was fine untill i installed the latest V2.11 .
did a comparision with v2.10C but i must have missed something.
Thank very much for your time and by the way, i love the station list, its super easy to use.

Here is the message error:

=================================================================================
---- Opened the serial port COM3 ----
E (1039) psram: PSRAM ID read error: 0xffffffff
[     3][W][esp32-hal-psram.c:71] psramInit(): PSRAM init failed!

ESP32 Chip: ESP32-D0WD-V3
Arduino Version: 2.0.11
ESP-IDF Version: 4.4.4
MiniWebRadio     Version 2.11 Aug 31/2023  
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 4194304 bytes, speed 40 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 80 MHz
PSRAM not found! MiniWebRadio will not work properly!
assert failed: xQueueSemaphoreTake IDF\components\freertos\queue.c:1545 (( pxQueue ))

Backtrace: 0x40084e29:0x3ffc7ad0 0x4008f735:0x3ffc7af0 0x40099afd:0x3ffc7b10 0x40090871:0x3ffc7c40 0x400d7837:0x3ffc7c80 0x400f1336:0x3ffc7ca0 0x400dc424:0x3ffc7cf0 0x4010b575:0x3ffc7d50

ELF file SHA256: e701809e218b501b

---- Opened the serial port COM3 ----
E (1039) psram: PSRAM ID read error: 0xffffffff
[     3][W][esp32-hal-psram.c:71] psramInit(): PSRAM init failed!

ESP32 Chip: ESP32-D0WD-V3
Arduino Version: 2.0.11
ESP-IDF Version: 4.4.4
MiniWebRadio     Version 2.11 Aug 31/2023  
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 4194304 bytes, speed 40 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 80 MHz
PSRAM not found! MiniWebRadio will not work properly!
assert failed: xQueueSemaphoreTake IDF\components\freertos\queue.c:1545 (( pxQueue ))

Backtrace: 0x40084e29:0x3ffc7ad0 0x4008f735:0x3ffc7af0 0x40099afd:0x3ffc7b10 0x40090871:0x3ffc7c40 0x400d7837:0x3ffc7c80 0x400f1336:0x3ffc7ca0 0x400dc424:0x3ffc7cf0 0x4010b575:0x3ffc7d50

Rebooting
==============================================================================
[Compiling V2.11.txt](https://github.com/schreibfaul1/ESP32-MiniWebRadio/files/12498055/Compiling.V2.11.txt)

schreibfaul1, 2023-09-02:
This is a strange error. The PSRAM is recognised before the Setup() is called. So a connection with the new version is unlikely. Is the board really running with an older version? On some boards, the connections between PSRAM and ESP chip are led out and cannot really be used as GPIOs. Maybe one of these pins is contacted after all?

What must not be is the resulting continuous loop and restart. I have now prevented this.

ve2cfb, 2023-09-02:
Hello Wolle,
Just to clarify things...
My three esp32 modules that i have tried are:  ESP-WROOM-32  (ESP32 DEVKIT V1)  4M ram and they are 30 pins modules.
I never had problem before. From your comment, it looks like i have external PSram but its not the case.

Many thanks for your time.... Pierre

ve2cfb, 2023-09-03:
it seem that my ESP32 EROOM does not  have a PSRAM. I tried a WROVER-E (8M PSRAM) and it worked ok.
I still dont know why my previous version did worked.  go figure...

schreibfaul1, 2023-09-04:
Hi @ve2cfb,
in earlier versions many things worked without PSRAM, but now the project is so big that a PSRAM is required. The GPIO2 of the ESP32 is a "strapping pin" I think it has to be grounded at boot time, but the SD card expects a pullup. The pullup is turned on by SW in Setup(). Sometimes the HW pullup is sufficient and the SD_MMC is no longer recognised. ILI9488 is indeed a problem case and does not manage 80MHz in my case. ILI9431, ILI9486 and ST7796 have no problems with 80mHz.

Rich1975, 2023-09-05:
Hi Wolle,
So, as per your last comment does it mean "upgrade your "hard ware" or find a different program",
Would be nice to have the earlier versions for those of us who didnt want those extra bells and whistles and if so how do we get the earlier versions?
Even so, thanks for your efforts, I do appriciate it, it has given me something to think about and looking forward to continuing to be involved in electronics says this old man

valera39, 2023-09-05:
[Rich1975](https://github.com/Rich1975)
Here's version 1.30a 
[ESP32-MiniWebRadio-master.zip](https://github.com/schreibfaul1/ESP32-MiniWebRadio/files/12526875/ESP32-MiniWebRadio-master.zip)

Rich1975, 2023-09-05:
Thank you valera39, I will give it a try, about 2 months ago everything seemed to be OK and was getting ready to build an enclosure for it and then it crashed, Thank you again.

valera39, 2023-09-05:
[Rich1975](https://github.com/Rich1975) Please, I still have saved older versions and libraries for the radio

Rich1975, 2023-09-06:
valera39,
Thank you for the link to earlier code, could I trouble you to explain how to use it with PlatfomIO, Here's what I did....1) down load the zip file, 2) extract all, 3) start PlatformIO, 4) Close Folder (because it may be something else,) 6) Click "Open Folder and choose the folder where I had extracted the zipped file. NEXT if I try to Build I get an error [.pio\build\esp32dev\src\main.cpp.o] Error 1.  This is at the top of the screen  MiniWebRadio -- Webradio receiver for ESP32, 2.8 color display (320x240px) and VS1053 MP3 module.  *
//*********************************************************************************************************
//
// first release on 03/2017
// Version 30a, Aug 18/2021
Have I provided enough info and could you help me?

valera39, 2023-09-06:
[Rich1975](https://github.com/Rich1975)
I compiled this version following Wolle's instructions in the additional_info folder.  I compiled it in Arduino IDE and it worked right away. And since the second version I used PlatformIO. Try to use Arduino IDE for the first version, you should definitely succeed, there is nothing complicated, just carefully read the instructions. The version I sent you is the last one that compiles also in Arduino IDE.  The next versions are only in PlatformIO.

MikeF59, 2023-09-23:
hi ,
I tried to compile v. 1.30a from above for esp32 with platformio and I get this error:

src/main.cpp:169:49: error: no matching function for call to 'VS1053::VS1053(int, int, int)' 
 VS1053 vs1053(VS1053_CS, VS1053_DCS, VS1053_DREQ);
                                                 ^
In file included from src/main.cpp:77:
.pio/libdeps/esp32dev/ESP32-vs1053_ext/src/vs1053_ext.h:279:5: note: candidate: 'VS1053::VS1053(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)'
     VS1053(uint8_t _cs_pin, uint8_t _dcs_pin, uint8_t _dreq_pin, uint8_t spi, uint8_t mosi, uint8_t miso, uint8_t sclk);
     ^~~~~~
.pio/libdeps/esp32dev/ESP32-vs1053_ext/src/vs1053_ext.h:279:5: note:   candidate expects 7 arguments, 3 provided
.pio/libdeps/esp32dev/ESP32-vs1053_ext/src/vs1053_ext.h:105:7: note: candidate: 'VS1053::VS1053(const VS1053&)'
 class VS1053 : private AudioBuffer{
       ^~~~~~
.pio/libdeps/esp32dev/ESP32-vs1053_ext/src/vs1053_ext.h:105:7: note:   candidate expects 1 argument, 3 provided

What is a problem, please ? Can You help me ?

closed 2024-12-08 by ve2cfb