schreibfaul1/ESP32-MiniWebRadio #304: StationList-Titles not shown and played on Display/Web

Robert131, 2023-10-12:
Hello Wolle,

i have edited the sd-card with 3 tites and added the dedicated logos. Then i have deleted all remaining titles. Uploading with Filezilla was fine but the new stations are not shown in the radios display.
On the drop-down menue from the web interface i can select the 3 titles but instead of the modified stations the old stations were displayed and selected.
All other functions works well but the handling with the station list seems for me not quite clear.

schreibfaul1, 2023-10-12:
Hello,
to edit the station list, it is best to go to the website. With the "Radio Browser", you can search for new stations and test the URL for functionality. Often a station logo is included. Otherwise, you can find a suitable logo using the Googe image search, for example. Then you transfer the URL of the image found and the image is scaled to fit. Each logo must be given the name of the station so that it can be assigned, then save the logo to the SD card with the button. Each new station is transferred to the list. The list is then saved. You can also save the list in EXCEL format for possible recovery on your PC.

Robert131, 2023-10-13:
Hello,
thank you for the clarification. With : Add new station to list in the "Radio Browser" then Upload to SD-Folder and following Save to SD in the "Radio-Tower" menü it works.

closed 2023-10-13 by Robert131