schreibfaul1/ESP32-MiniWebRadio #308: feature request > BLUETOOTH SPEAKER

clear-sky, 2023-10-20:
Hi dear wolle!
all esp32 cores have bluetooth i am wondering why anyone does not have any plan to use it!
in this project adding bluetooth speaker does not need additional hardware only needed software
what is your opinion?

trench29, 2023-10-22:
Hi! I would also be happy about the bluetooth extension. Currently I use two different music devices for playback. An internetradio and a bluetooth speaker. If both functions can be combined into the ESP32-MiniWebRadio, then I would only need one device. That would be great! :)

steve6375, 2023-10-25:
I think esp32 has a limitation in that you can have WiFi or Bluetooth but not both.
If this is the case then only playing of media files from flash drive via Bluetooth would be possible?

clear-sky, 2023-10-26:
Hi dear steve6375!
no problem if that limitation exist!
in bluetooth speaker mode no need to wifi and in wifi radio mode no need to bluetooth
but this is not possible to play wifi radio (or mp3 file) and send it to another bluetooth speaker 
if you know edzelf he has solved this problem with an external bluetooth sender!
https://github.com/Edzelf/ESP32Radio-V2

schreibfaul1, 2023-10-28:
It's not that easy, and there are some limitations
- the ESP32-S3 can't do BT classic 
- it doesn't work with a VS1053 board 
- I would have to lock the IRAM functionality for WiFi and LWIP, because much IRAM has to be reserved for BT 
- the project already needs so much flash memory that BT doesn't fit in 4MB anymore

clear-sky, 2023-10-29:
Hi dear wolle !
 the facts that you said means that is impossible!
**thanks with respect**

closed 2023-10-29 by clear-sky

schreibfaul1, 2023-10-29:
This is not impossible. The flash is now filled to 80%, with BT classic 105% is needed. That functions then only with 8MB. Wait and see what is needed with IDF5.1.

clear-sky, 2023-10-29:
Hi dear wolle!
this is only a suggest
you can add additional patch for some users that can use it - for example BT classic for esp32 with 8 or 16 Mb flash

valera39, 2023-10-29:
Dear Wolle!
I think bluetooth is a completely unnecessary feature. It would be better to further develop and improve the media player and full DLNA support
Thank you for your hard work.

schreibfaul1, 2023-11-03:
For the ESP32 I have implemented BT as A2DP Sink (BT Speaker). To get the necessary flash memory I deactivated the Ethernet function (in menuconfig). I don't think anyone will use an Ethernet adapter. The ESP32-S3 does not offer BT Classic, only BLE. BLE from V5.2 can support audio but is currently not supported by Espressif. The extension is in the Arduino V3.0 branch. Several symbols must be copied to the SD card, it is best to replace the entire contents of the btn and common folders.

The BT connection must always be established from the audio source. The first BT connection requires a PIN, the default PIN is 1234.
    
[image]
If the connection is successful, the logo changes from silver to gold. [image]
If metadata is transmitted, ARTIST and TITLE are extracted from it and shown on the display.
The functions of the buttons on the left and right depend on the application of the audio source and can therefore vary. The orange bars in the footer indicate the BT signal strength. BT is currently not supported via the website.

clear-sky, 2023-11-04:
Hi dear wolle!
thanks for your great work-this project is going to be famous for all over the world!
unfortunately i am on esp32-s3 and can not test and most wait for 5.2!
on the other hand is any good news form usb host for esp32-s3?
_thanks with respect_

Arne65, 2023-11-10:
Hallo Wolle

ich habe das auf einem  ESP32-DevKitC-VIE  (Arduino 3.0) probiert den BT SD Container auf die SD Karte soweit so gut.

Ich bekomme aber keine BT Verbindung zu Stande an meinem Endgerät (IPhone) müsste doch eigentlich der esp zu sehen sein .Oder ? Da die Kopplung ja vom Endgerät aus geht.

Was mache ich falsch ? Oder habe ich das was grundlegend falsch verstanden .

schreibfaul1, 2023-11-10:
Eigentlich nichts. Wenn du den BT Modus geöffnet hast und das Logo silbern ist wird ein "Advertising" gestartet. Dein Smartphone wird dann das neue Gerät als "MiniWebRadio" erkennen. Bei der Kopplung wird eine Pin (1234) verlangt. Wenn das Gerät gekoppelt ist, wechselt das Logo auf Gold. 
Ich habe das mit verschiedenen Smartphones, einem Tablet und einem Linux-PC getestet. Das hat immer funktioniert.

Arne65, 2023-11-10:
Ich glaube mein Problem liegt an dem BT Modus .Ich habe auf dem Display nirgendwo einen Button um um den zu starten .Wie komme ich denn da hin . 

Danke für deine schnelle Antwort und Mega Respekt für deine  großartigen arbeitet die du hier leistest.

schreibfaul1, 2023-11-10:
Ich denke, der wird da sein, du siehst ihn nur noch nicht. Ich habe für BT einen neuen Button gebraucht. Lade bitte von der Conten_on_SD_Card.zip die Symbole auf die SD Karte. Da das mehrere sind, ist es besser den ganzen btn und common Ordner zu kopieren.
Dann kannst du auch vom Touchscreen die Quelle steuern. Z.B.mit dem VLC Player funktioniert das, Einzel oder Doppelklick (ausprobieren)

Arne65, 2023-11-10:
Die beiden Ordner inclusive der BT Button sind alle auf der SD Karte hatte die zip unter dem Arduino Zweig geladen und auf die Karte gepackt

Arne65, 2023-11-10:
Allerdings sind in dem ZIP Ortner nur grüne Blau und rot 
Silber und Gold Abt Button sind da nicht

Arne65, 2023-11-10:
Sorry unter Common sind die da

schreibfaul1, 2023-11-10:
Hab eben nochmal geguckt. Die sind da. z.B. BTnc.jpg und BT.jpg sind die Logos.
Das ist nicht die zip aus dem Master, du benötigst die aus dem Branch ArduinoB3.0

Arne65, 2023-11-10:
Ja die aus dem Branch Arduino-V3.0 habe ich da sind ja auch alle Button und auch BT RSSI 

komisch ist eigentlich sollte ja auch unten der BT RSSI Indikator sein der auch nicht da ist .

Den Code habe ich aus dem Arduino 3.0 Branch das ist doch sicher richtig

schreibfaul1, 2023-11-10:
Absolut, das ist merkwürdig. Der entscheidende Button ist BT_Green.jpg
Der sollte kommen wenn du das Display 2x berührst, rechts neben der Lampe für die Helligkeit

Arne65, 2023-11-10:
Hmm also rechts neben Helligkeit passiert nichts 

rechts neben Station List bekomme ich invalid Button 6 /7/8 aber ich denke das wird normal sein eventuell als Platzhalter für weitere Button

Arne65, 2023-11-10:
Wenn ich ganz rechts neben Helligkeit 2 X tippe 

stürzt der ESP ab 

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008cf59  PS      : 0x00060f30  A0      : 0x800d5161  A1      : 0x3ffd0150  
A2      : 0x0ef65020  A3      : 0x0ef6501c  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffd0110
A10     : 0x3ffba414  A11     : 0x3ffba414  A12     : 0x00000004  A13     : 0x00060d23  
A14     : 0x00060d23  A15     : 0x00000001  SAR     : 0x00000007  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0ef65020  LBEG    : 0x4008cf59  LEND    : 0x4008cf69  LCOUNT  : 0xffffffff  

Backtrace: 0x4008cf56:0x3ffd0150 0x400d515e:0x3ffd0160 0x400d59a0:0x3ffd0180 0x400d5a01:0x3ffd01a0 0x400d6fde:0x3ffd01e0 0x400dc13a:0x3ffd0200 0x4011b4fe:0x3ffd0220 0x400dce7f:0x3ffd0240 0x4012ec69:0x3ffd02a0

  #0  0x4008cf56:0x3ffd0150 in ?? ??:0
  #1  0x400d515e:0x3ffd0160 in ?? ??:0
  #2  0x400d59a0:0x3ffd0180 in ?? ??:0
  #3  0x400d5a01:0x3ffd01a0 in ?? ??:0
  #4  0x400d6fde:0x3ffd01e0 in ?? ??:0
  #5  0x400dc13a:0x3ffd0200 in ?? ??:0
  #6  0x4011b4fe:0x3ffd0220 in ?? ??:0
  #7  0x400dce7f:0x3ffd0240 in ?? ??:0
  #8  0x4012ec69:0x3ffd02a0 in ?? ??:0

ELF file SHA256: 6b5c93e3fbf846fd

Rebooting...
ets Jul 29 2019 12:21:46

schreibfaul1, 2023-11-10:
Okay, dann setzt das Board nicht CONFIG_IDF_TARGET_ESP32
[image]
wird bei dir die untere Zeile aktiv sein. Du hast bestimmt das 8MB board in platformio ausgewählt. Wenn das so ist, nimm mal bitte das 4MB Board

Arne65, 2023-11-10:
Hab das dann mal mit 4MB Board probiert abstützen tut er jetzt nicht aber wie schon schrieb passiert da rechts neben Helligkeit  leider nichts

schreibfaul1, 2023-11-10:
Ich habe keine weitere Idee. Es wird nur zwischen ESP32 und ESP32-S3 unterschieden. Ist es ein ESP32 wird der BT Button angezeigt. Bevor du weiter suchst, würde ich den Ordner löschen, das Repository mit VS Code neu klonen , den Arduino Branch links unten auswählen und die Einstellungen in common.h neu setzen.

Arne65, 2023-11-11:
Habe dann nochmal neu  gemacht und siehe da es läuft perfekt.

Vielen Dank 
 Eine  kleine Frage die betrifft aber nicht BT.

Könntest Du die vorstellen MQTT zu integrieren ?