schreibfaul1/ESP32-MiniWebRadio #309: feature request > ADD IR BUTTONS

clear-sky, 2023-10-21:
Hi dear wolle!
some users like to control your radio with remote completely
in this condition it is better to use a remote that have more keys like : menu exit up down left right ok etc

schreibfaul1, 2023-10-22:
At the moment I am preparing the SW for the Arduino V3 (IDF 5.1) including all necessary libraries. After that I can add more IR functions.

clear-sky, 2023-10-22:
thanks  for the reply!

clear-sky, 2024-05-22:
Hi Dear Wolle
I decided to add ir bottons but found some problems
i intend to increase ir bottons to 50 and can not correct some functions if possible please help

line 575 from main.cpp

```
void saveIRbuttonsToNVS() {
    uint8_t  ir_addr = ir.get_irAddress();
    uint8_t* ir_buttons = ir.get_irButtons();
    char     buf[12];
    pref.putShort("irAddress", ir_addr);
    for(uint8_t i = 0; i < 20; i++) {
        sprintf(buf, "button_%d", i);
        pref.putShort(buf, ir_buttons[i]);
        log_i("i=%i ir_buttons[i] %X", i, ir_buttons[i]);
    }
    pref.putShort("IR_numButtons", 20);
}

void loadIRbuttonsFromNVS() {
    // load IR settings from NVS
    uint numButtons = pref.getShort("IR_numButtons", 0);
    ir.set_irAddress(pref.getShort("irAddress", 0));
    char    buf[12];
    uint8_t cmd = 0;
    for(uint i = 0; i < numButtons; i++) {
        sprintf(buf, "button_%d", i);
        cmd = pref.getShort(buf, 0);
        ir.set_irButtons(i, cmd);
    }
}
```

line 3278 from main.cpp

```
    if(cmd == "loadIRbuttons"){    loadIRbuttonsFromNVS(); // update IR buttons in ir.cpp
                                   char buf[150];
                                   uint8_t* buttons = ir.get_irButtons();
                                   sprintf(buf,"0x%02x,", ir.get_irAddress());
                                   for(uint8_t i = 0; i< 20; i++){
                                        sprintf(buf + 5 + 5 * i, "0x%02x,", buttons[i]);
                                   }
                                   buf[5 + 5 * 20] = '\0';
                                   webSrv.reply(buf, webSrv.TEXT); return;}
```

schreibfaul1, 2024-05-25:
Hi clear-sky, 
that won't be enough, the remote control code is stored in NVS for each button. Then the button must be assigned a function. This is currently lacking. One idea might be to assign buttons to several functions, such as audio player, DLNA or ...

clear-sky, 2024-08-07:
Hi Dear Wolle!
Have you enough time to add ir buttons?

schreibfaul1, 2024-08-08:
Hi clear-sky,
there are enough ideas. Now I want to fix the bugs after the last update, the new AAC decoder does not always work. I would like to extend the OPUS decoder so that streams work in hybrid mode.
The IR will not continue until the end of fall.

valera39, 2024-08-08:
If possible, add at least one on/off button. At the same time, keep the ability to turn the radio off by long pressing the "mute" button so that you can turn the radio off with a remote that does not have a separate on/off button.

closed 2025-03-31 by github-actions[bot]