schreibfaul1/ESP32-MiniWebRadio #310: Feature request for Play all files and Random.

ve2cfb, 2023-10-23:
Hello Walle,

Your files listing is  working  very good and the play all files is also great.
I did a basic test to see if i was able to have the RANDOM feature when playing all files but  i am still looking in the Button section and 
 i'm not an expert yet.. 

Here is the code... starting at line 2287 in main.cpp
==================================================================
    if(!_f_playAllFiles) {
        if(audioFile) audioFile.close(); // maybe audioFile contains old data
        
        _cur_AudioFileNr = random(0 , _SD_content.size()); //====================  Added
        _curAudioFolder = folderPath;
        _cur_AudioFileNr = random(0 , _SD_content.size()); //====================  Added
       
        _f_playAllFiles = true;
        changeState(PLAYERico);
        showVolumeBar();
===================================================================

In the "Audio player" menu, it would be nice to add an extra "random button" and possibly make the "random button " and the "play all file button" like a latch button  (Red or Green).  

Thanks for your time and have a nice day.... Pierre

steve6375, 2023-10-23:
Actually we do not want a random feature - we want a random shuffle (sort) feature.

If random is used then it could play the same song twice consecutively (A H **J J** G D) or more frequently - e.g.  **A** H J **A** S K **A**

I believe the same thing happened with early MP3 players. The users asked for a random feature but then they complained because it would often play the same tune twice or too often - what they really wanted was a random shuffle feature.

ve2cfb, 2023-10-23:
Very interesting. I did notice that especially on small number of files, the song came more often.
Is it possible to integrate this feature in MiniWebRadio... ?

Many thanks for the info.

hengist60, 2023-10-24:
I think the issue here is that generating a sequence of random numbers where there is no restriction on the likelihood of the same number reappearing more than once (“sampling with replacement”) is relatively easy (technique first developed by Bletchley Park codebreakers post WWII).  But for “sampling without replacement”, the algorithm needs to somehow keep a record of what has already been sampled and prevent it appearing in the ‘random sequence’ again. 

Such “shuffling” is  commonly required  in statistical packages and  - as I understand it - this is usually done by having a second vector of “weights”: initially a vector of 1s. Once sampled an “observation” (in this case a musictrack) is given a weight of 0 which is then used to prevent it appearing in the ‘shuffled vector” again (eg. https://www.rdocumentation.org/packages/base/versions/3.6.2/topics/sample). With an Arduino Mega I already use https://github.com/pmjdebruijn/Arduino-Entropy-Library/blob/master/Entropy.h  which works much better than the Arduino random()  as a MP3 player “random track generator” but I don’t care about the same track appearing more than once.

ve2cfb, 2023-10-24:
Hello Jon,
Thanks for the info. i will do some research and basic test with entropy. 
I like what you said  **_"but I don’t care about the same track appearing more than once"_**
In my case, all my MP3 directory on the SD Card contain at least 50 mp3 files each, so it is not a big deal if one song is played again. To be followed...

Random was only a initial test but my other concern was the following... i beleive it can reduce the number of operations.
One drawback, is when you either power on the unit or switch from Radio Mode to Audio Mode, you are pointing to the same music directory. That is OK but if i want to chage my directory, i have to flip back and forth .

Hope my text will explain a bit better....

- 1- from the **_Internet Radio_** Menu, go to **_Audio Player_** menu.
- 2- Add an extra **_random/shuffle button_**
- 3- make **_play_all button_** and **_random button_**  only latchable with blue/red color. Simple switch.
- 4- Then, when going to **_file list button_** , we can change to a new mp3 directory, pick a mp3 song, and automaticlly it goes  back to playing the song in **_random/shuffle mode_** or **_all files mode_**.

**Many thanks to Wolle and all the others who help in building and participating in this wonderful project.**

Pierre

steve6375, 2023-10-25:
https://cpp.hotexamples.com/examples/-/-/playlist_shuffle/cpp-playlist_shuffle-function-examples.html

schreibfaul1, 2023-10-26:
Hi,
I have implemented the function, but currently not in the master, but in the branch Arduino-V3. For testing, it is not enough to select the branch in VS Code, new symbols must also be copied to the SD card, player.gif and Button_Shuffle_Blue[Yellow]. jpg

[image]

ve2cfb, 2023-10-26:
Hello Wolle,
Super...
 Thank you very much.

closed 2023-10-26 by ve2cfb