schreibfaul1/ESP32-MiniWebRadio #312: Issue with ILI9341 touch coordinates

Rhombur, 2023-11-01:
Hi,
I had a little issue with the touchpad, which in principal worked but the buttons and the touch location were not matching.
This could be traced-down to the coordinate calculations in tft.cpp, starting with line 4877. Using the changed code below both rotations work with my display; yet I do not know if I just get a crude one from China or if this is a general issue and something got mixed-up.
Beside this little issue, the whole things works perfectly fine, even if working for the first time with Platform.io 
Many thanks!
 
```
//-------------------------------------------------------------
    if (TP_vers == 0) {  // 320px x 240px
        if (_rotation == 0) {
            y = ILI9341_HEIGHT - y;
        }
        if (_rotation == 1) {
            tmpxy = x;
            x = y;
            y = tmpxy;
            y = ILI9341_WIDTH - y;
            x = x;
        }
        if (_rotation == 2) {
            x = ILI9341_WIDTH - x;
        }
        if (_rotation == 3) {
            tmpxy = x;
            x = ILI9341_HEIGHT - y;
            y = tmpxy;
        }
    }
```

steve6375, 2023-11-01:
Depends on what you set TP_vers to really, doesn't it?

schreibfaul1, 2023-11-01:
Hello Rhombur,
please try first, as Steve recommended, another TP, e.g. the HX8347D. Rotation of TFT and TP do not have to be the same, but are almost always 1 or 3.
If you have an "exotic" display, you can determine the values for Xmax, Xmin, Ymax, Ymin yourself. The line in readTP() ` log_w("_x %i, _y %i", _x, _y);` shows you the value. This is the voltage value in mV that the TP controller determines. And at the bottom `log_i("TP_vers %d, Rotation %d, X = %i, Y = %i",TP_vers, _rotation, x, y);` shows the calculated coordinate. You do this in the upper left corner and in the lower right corner.

Rhombur, 2023-11-03:
Hi,
thanks. I checked the coordinates from the serial monitor, which let me do the changes shown above. However, it looks like the HX8347D TP also works without the need of modifications ... seems my assumption that an ILI9341 links with the same TP was too optimistic ;)
I have to find time for some more trials, but first need to wait for a more powerfull ESP ... the model I have seems to work on the limit, taken the slow response to the screen.

steve6375, 2023-11-03:
To improve response time, try increasing the frequency settings until you reach the maximum frequency that it will work reliably at 
Also tweak SD card frequency, etc.

Arne65, 2023-11-07:
Hello

If your display reacts a bit slowly you can try setting #define AUDIOTASK_CORE 1 // 0 or 1 to 0 in Common.h.

There is a small disadvantage with streams with 256 K, there are drops at the beginning (about 2 seconds).

closed 2025-02-15 by github-actions[bot]