schreibfaul1/ESP32-MiniWebRadio #313: Issue with ESP32-S3-WROOM-1 N16R8

JozefKc, 2023-11-11:
Hi,
I am a newbee in Platformio a I would like to ask you about this problem:
I tried to programm actual version of the radio on **ESP32-S3-WROOM-1 with N16R8** and received this:

**Display:**
WiFi credentials are not correct
Accesspoint IP:
192.168.4.1

**Terminal:**
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
E (518) psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (518) spiram: SPI RAM enabled but initialization failed. Bailing out.
[0;32m[   539][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled[0m
[0;32m[  1190][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: -3-[0m
[0;32m[  1203][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: -1-[0m
[0;32m[  1204][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: -3-[0m
[0;32m[  8291][I][WiFiMulti.cpp:114] run(): [WIFI] scan done[0m
[0;32m[  8292][I][WiFiMulti.cpp:119] run(): [WIFI] 14 networks found[0m
[0;32m[  8293][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: F0:79:59:D1:25:4C SSID: -3- Channel: 13 (-45)[0m
[0;33m[ 11438][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT[0m
[0;31m[ 13336][E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (6).[0m
[0;33m[ 13387][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 8 - ASSOC_LEAVE[0m
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

I tried to connect to the -3- or -1- network with the correct password - result is: not connected.
I use ILI9341 display with SPI v1.2 and touch HSD028309 E4. Display works fine, but the touch does not - reacts on some touches,
but only in some places and the system after some touches makes a reset...

Now I do not know how to solve this. I will be grateful for any advice.

Jozef

P.S.: I tried to make a simple programm to get time from NTP trough my Wifi - everything worked fine....

Arne65, 2023-11-11:
Hallo 

Did you change the PSRAM to octo under menuconfig?

JozefKc, 2023-11-11:
Sorry. Which file file it is? I did not change it.

Arne65, 2023-11-11:
The esp32s3 16R8 has a different PSRAM, which you have to adjust under menuecofig. Check here 

issues PSRAM size not printed in Serial Monitor. #300

In platformio.ini it says at the bottom under *1 how you get to the relevant point.

Since your PSRAM is not recognized, this could possibly be your problem. You have to switch it to octal.

JozefKc, 2023-11-11:
In ESP32-S3-DevKitC-1-N16R8.json is **opi**:

 "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_16MB.csv",
      "memory_type": "qio_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_ESP32S3_DEV",
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_USB_MODE=1",
      "-DARDUINO_USB_CDC_ON_BOOT=1"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "psram_type": "**opi**",
    "hwids": [
      [
        "0x303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi",
    "bluetooth"
  ],
  "debug": {
    "default_tool": "esp-builtin",
    "onboard_tools": [
      "esp-builtin"
    ],
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Espressif ESP32-S3-DevKitC-1-N16R8V (16 MB QD, 8MB PSRAM)",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 16777216,
    "require_upload_port": true,
    "speed": 921600
  },
  "url": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html",
  "vendor": "Espressif"
}

**and in platformio.ini:**

        env:esp32]
        board_build.partitions = miniwebradio4MB.csv    ; 4MB Flash
        ;board_build.partitions = miniwebradio8MB.csv    ; 8MB Flash
        ;board_build.partitions = miniwebradio16MB.csv   ; 16MB Flash
        board = ESP32-Dev-4MB                   ; 4MB Flash  *2)
        ;board = ESP32-Dev-8MB                   ; 8MB Flash  *2)
        ;board = ESP32-Dev-16MB                  ;16MB Flash  *2)
build_flags =
        -DCORE_DEBUG_LEVEL=3                    ; 0 None, 1 Error, 2 Warn, 3 Info, 4 Debug, 5 Verbose
        -DCONFIG_ARDUHAL_LOG_COLORS
        -DBOARD_HAS_PSRAM
        -DCONFIG_IDF_TARGET_ESP32
        -DARDUINO_RUNNING_CORE=1                ; Arduino Runs On Core (setup, loop)
        -DARDUINO_EVENT_RUNNING_CORE=1          ; Events Run On Core
build_unflags =
        -DCONFIG_IDF_TARGET_ESP32S3
        -Wall
        -Wextra
        -DARDUINO_RUNNING_CORE                  ; Arduino Runs On Core (setup, loop)
        -DARDUINO_EVENT_RUNNING_CORE

[env:esp32s3]
        ;board_build.partitions = miniwebradio4MB.csv    ;  4MB Flash
        ;board_build.partitions = miniwebradio8MB.csv    ;  8MB Flash
        board_build.partitions = miniwebradio16MB.csv   ; 16MB Flash

        ;board = esp32-12k                      ;  4MB Flash, 8MB quad PSRAM
        ;board = ESP32-S3-DevKitC-1-N8R2        ;  8MB Flash, 2MB quad PSRAM
        ;board = ESP32-S3-DevKitC-1-N8R8        ;  8MB Flash, 8MB octo PSRAM  *1)
        **board = ESP32-S3-DevKitC-1-N16R8       ; 16MB Flash, 8MB octo PSRAM  *1)**

Arne65, 2023-11-11:
you have to change the PSRAM to octal under run menueconfig

JozefKc, 2023-11-12:
Many thanks for your help. I finally found what it was. I had a bit of a problem with how to switch between quad and oct, but in the end I don't even know how to switch. After saving, translating and uploading, the radio is running, but I have to solve the problems with the touch surface and the distortion of the icons in the lower part of the screen. But all problems will be solved gradually.
If anyone needs it, I am sending the json configuration for ESP32-S3-WROOM1-N16R8 in the following comment.
Many thanks again.

JozefKc, 2023-11-12:
File for **ESP32-S3-WROOM1-N16R8** named as ESP32-S3-DevKitC-1-N16R8.json:

{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_16MB.csv",
      "memory_type": "qio_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_ESP32S3_DEV",
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_USB_MODE=1",
      "-DARDUINO_USB_CDC_ON_BOOT=1"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "psram_type": "opi",
    "hwids": [
      [
        "0x303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi",
    "bluetooth"
  ],
  "debug": {
    "default_tool": "esp-builtin",
    "onboard_tools": [
      "esp-builtin"
    ],
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Espressif ESP32-S3-DevKitC-1-N16R8V (16 MB QD, 8MB PSRAM)",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 16777216,
    "require_upload_port": true,
    "speed": 921600
  },
  "url": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html",
  "vendor": "Espressif"
}

Arne65, 2023-11-12:
This is my configuration in common.h for the ILI9341 display.

#define DECODER                      1                                  
#define TFT_CONTROLLER         0                            
#define DISPLAY_INVERSION    0                               
#define TFT_ROTATION              3                               
#define TFT_FREQUENCY          40000000                       
#define TP_VERSION                1                              
#define TP_ROTATION              1                               
#define AUDIOTASK_CORE       0                               
#define AUDIOTASK_PRIO        2           

You can also use TFT FREQUENCY 27000000

For me, the ILI9341 with the above configuration works very quickly and without any problems

JozefKc, 2023-11-14:
Radio works. The main problem now is that the touch is inverted - it means that buttons showed at the bottom do react by touching the on the upper side of the screen. And rotation does not help, because I need mirroring of touch by x - axe or y-axe. Any one of this would help. I can not find where to change it /looking for the lib for TP-sensing or somewhere else/. Any sugestion?
[image]

steve6375, 2023-11-14:
https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/313#issuecomment-1807207290

So these TP settings don't work with same TFT_ROTATION setting?

valera39, 2023-11-14:
[JozefKc](https://github.com/JozefKc) Try various combinations of tft_rotation and tp_rotation, I had the same problem

JozefKc, 2023-12-02:
Rotation does not help. Problem solved:
in tft.cpp you have to modify row nr.4920 in that way:

//-------------------------------------------------------------
    if (TP_vers == 2) {  // 320px x 240px
        if (_rotation == 0) {
            ;  // do nothing
        }
        if (_rotation == 1) {
            **tmpxy = x;
            x = y;
            y = tmpxy;**
            if (x > HX8347D_HEIGHT - 1) x = 0;
            if (y > HX8347D_WIDTH - 1) y = 0;
        }
        if (_rotation == 2) {
            x = HX8347D_WIDTH - x;
            y = HX8347D_HEIGHT - y;
            if (x > HX8347D_WIDTH - 1) x = 0;
            if (y > HX8347D_HEIGHT - 1) y = 0;
        }
        if (_rotation == 3) {
            tmpxy = y;
            y = x;
            x = HX8347D_HEIGHT - tmpxy;
            if (x > HX8347D_HEIGHT - 1) x = 0;
            if (y > HX8347D_WIDTH - 1) y = 0;
        }
    }

JozefKc, 2024-01-03:
If you use **ILI9341 with touch** /https://www.aliexpress.com/item/1005005378490219.html?pdp_npi=4%40dis%21EUR%21%E2%82%AC%205%2C96%21%E2%82%AC%205%2C24%21%21%21%21%21%402101fb0d17042905786086805ebf1d%2112000036815317889%21sh%21SK%21709507339%21&spm=a2g0o.store_pc_allItems_or_groupList.new_all_items_2007585817841.1005005378490219/ , this display is compatible with **HX8347D** touch driver. It's orientation is a bit strange. For the correct use I had to define it in **common.h** :
#define TFT_CONTROLLER      0                               // (0)ILI9341, (1)HX8347D, (2)ILI9486a, (3)ILI9486b, (4)ILI9488, (5)ST7796, (6)ST7796RPI
#define DISPLAY_INVERSION   0                               // (0) off (1) on
#define TFT_ROTATION        3                               // 1 or 3 (landscape)
#define TFT_FREQUENCY       40000000                        // 80000000, 40000000, 27000000, 20000000, 10000000
#define TP_VERSION          2                               // (0)ILI9341, (1)ILI9341RPI, (2)HX8347D, (3)ILI9486, (4)ILI9488, (5)ST7796, (3)ST7796RPI
#define TP_ROTATION         3                               // 1 or 3 (landscape)

and in **tft.cpp** starting row 4920:

    //-------------------------------------------------------------
    if (TP_vers == 2) {  // 320px x 240px
        if (_rotation == 0) {
            ;  // do nothing
        }
        if (_rotation == 1) {
            tmpxy = x;
            x = y;
            y = tmpxy;
            if (x > HX8347D_HEIGHT - 1) x = 0;
            if (y > HX8347D_WIDTH - 1) y = 0;
        }
        if (_rotation == 2) {
            x = HX8347D_WIDTH - x;
            y = HX8347D_HEIGHT - y;
            if (x > HX8347D_WIDTH - 1) x = 0;
            if (y > HX8347D_HEIGHT - 1) y = 0;
        }
        if (_rotation == 3) {
          tmpxy = x;
          x = HX8347D_HEIGHT - y;
          y = HX8347D_WIDTH - tmpxy;
          if(x > HX8347D_HEIGHT - 1) x = 0;
          if(y > HX8347D_WIDTH - 1) y = 0;
        }
    }

closed 2025-03-01 by github-actions[bot]