schreibfaul1/ESP32-MiniWebRadio #314: Network connection

mbarcowboy, 2023-11-13:
Using ESP32-S3-WROOM-1 ( N16R8 )
Changed PSRAM to octal as per install instructions
Compiled and loaded without issues
on boot:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
[   989][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[  1652][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: MbarRadioNet
[  7922][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  7923][I][WiFiMulti.cpp:119] run(): [WIFI] 4 networks found
[  7924][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: AC:15:A2:97:93:3B SSID: MbarRadioNet Channel: 9 (-42)
[  9882][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 11821][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 12965][E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (6).
[ 13013][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 202 - AUTH_FAIL
[image]

The AP does not show up on my phone as available networks - so cannot connect to 192.168.4.1
Help please

Arne65, 2023-11-13:
You may have deactivated 2.4 GHz WiFi on your phone and only work with 5 GHz ?

mbarcowboy, 2023-11-13:
> You may have deactivated 2.4 GHz WiFi on your phone and only work with 5 GHz ?

No. 2.4GHz WiFi is active. I am able to connect to several other Esp8266 devices on my network.

schreibfaul1, 2023-11-13:
If the WiFi access data is not known, an AP is established. Go to Settings on your smartphone, temporarily disconnect from your WiFi router and connect to MWR. You can then enter the IP 192.168.4.1 in the Internet browser.

mbarcowboy, 2023-11-13:
I have tried doing that. The MWR AP does not show up as an available network.

schreibfaul1, 2023-11-13:
You can also enter the WiFi access data in common.h at the top by hand

mbarcowboy, 2023-11-13:
Done that.
#define _SSID               "MbarRadioNet"                        // Your WiFi credentials here
#define _PW                 "XxxxXxxxXxxx"                // changed for this post
#define DECODER             1                               // (0)VS1053 , (1)MAX98357A PCM5102A CS4344... (2)AC101, (3)ES8388, (4)WM8978

I have also entered my SSID and password into the networks.csv file with tab delimiters.

steve6375, 2023-11-13:
maybe password is incorrect? case? length? special characters?

mbarcowboy, 2023-11-13:
Checked, double checked and triple check. However that does not explained why the MWR AP does not show up when normal authentication fails.

mbarcowboy, 2023-11-14:
Tried the Simple WiFi Radio from the examples folder of the I2S library. No success!
Think I will go and get another board, or 2. Report back after test new HW.

JozefKc, 2023-11-14:
It is a quite strange thing. I am using the same board and there is no problem with wifi........

mbarcowboy, 2023-11-24:
Update:
Still waiting for ESP32-S3 N16R8 to arrive from China. Must be on a slow boat from China!
In the meantime, I managed to get a ESP32-S3 N8R2 and ESP32-Wrover-e 8MB locally. Both of these are working correctly and have good sound quality.  I may not bother with the N16R8s when they arrive! But then, I probably will.
PS
This is my first time using VSC/PIO - what a learning curve.

ovidiu169, 2023-12-22:
Hi, mbarcowboy 

I have the same problem - "WiFi credentials are not correct". Tried everything possible on this world. I tripled check the credentials, all is OK. I tried to access the Hotspots on my iPhone, my Lenovo tablet (of course I changed the credentials in "common.h file accordingly) - nothing worked. I changed different boards, including ESP32 S-3, nothing changed. The serial monitor says the the same no matter what i do - "AUTH_EXPIRED" - which is a non-sense because the credentials are not expired. I erased every time the flash containing NVS file and after I uploaded the code again. As in your case I tried to access IP 192.168.4.1.  It cannot be accessed. Nope, nothing, niente, nada...

schreibfaul1, 2023-12-22:
You can delete everything in main.cpp and replace it with an example from Espressif:
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiMulti/WiFiMulti.ino Is the WiFi connection OK then?

ovidiu169, 2023-12-23:
Hi Wolle,I tried but with no positive result. I tried also the new version of code, but I have also other errors…Trimis de pe iPhone‑ul meuPe

closed 2025-03-01 by github-actions[bot]