schreibfaul1/ESP32-MiniWebRadio #315: Problems with ESP32 DevKit Board

ovidiu169, 2023-11-21:
Hello Wolle !

I am trying hardly to reproduce your magnificent MiniWebRadio. It is a masterpiece of programming !
I used for the first time Platformio IDE. I've been documented a lot about this complex platform, but I am still a newbee...
So... I tried to use one of already bought ESP boards - PLUSIVO E 32 - made in China, of course. This is a ESP32 DEVKIT 1 mainly, with a flash of 4 M.
After many attempts I succeed to initiate the program(with no "SD Card Mount failed" message). As I saw already this not possible without pulling resistor on D0(10K). Why ? I don't know !... You said "no pull-up resistors on SD Card". This doesn't work for me.
I reduced the frequency on the card to 40000000, but still not working without pull-up resistor.
In the next stage of my "experiments" I tried to control the radio using touch screen(ILI9341). No way , because the virtual buttons doesn't appear on the screen !
Taking into consideration that it would be possible to not have enough PSRAM (or isn't detected) I mounted a push button in series with the pull-up resistor on the SD card. The operation became like that:
- First I press the button and connect power supply;
- after the program initialize and the image on the screen is dowloaded, I release the button 
- Touch the superior part of the screen, the base line appear;
- touch with a pen "invisible" buttons and select the stations, volume, etc...
It's like "hide and seek" , but it's all I got. I tried different boards - ESP32 DevKit C, ESP32 WROOM 32 D... The result is the same. 
I ordered a ESP32-DEVKIT C-1-N8R8 which has to come today. It is my final try.
Also, I did not succeed to find and open the MiniWebRadio app in internet. I searched everywhere in the Community Radio Browser, and also I searched through/using the IP Address displayed on the bottom of the screen(in Chrome) but no result.
I would be very grateful If You will be more explicit in using these metods of controlling the Radio...
Thank You in advance !!!

schreibfaul1, 2023-11-21:
Hello @ovidiu169, it looks like there are two basic problems: 
1) SD card
The SD card is operated as SD_MMC, connected as in the circuit diagram, i.e. all pins of the SD are connected directly to the ESP32. If you use a card adapter, e.g. of the display, there are certainly series or parallel resistors. These must be removed. I assume you have the "red" display, in which case the resistors R1, R2 and R3 are short-circuited or replaced by bridges.
2)PSRAM
A value greater than or equal to 2MB should be displayed in the log after booting.
[image]
If you are using the ESP32-S3 N8R8, the connection must be set to "octal" in menuconfig. Otherwise you will have a "reboot loop"

ovidiu169, 2023-11-21:
Thank you for your quick response!
Yes, I have the “red” display, indeed. But I did not used the socket on it, I conected the SD card adapter separately to GPIO 2, 15,14,3v3 and ground.
Maybe I should try to use the socket on the red display?!?
Second. Oh, yes I am fully aware that I have to change the settings of the ESP32-S3 from Quad to Octal. But I have no ideea how to do it…🫣

schreibfaul1, 2023-11-21:
Using the SD card adapter of the display is a good idea, as long as you bridge the series resistors. Especially with the ESP32, D0 (GPIO2) is a "strapping pin" which set a pull-up resistor via SW in Setup() after the boot process.

ovidiu169, 2023-11-21:
Yes, I got it ! I will try this, of course.
I just received the DAC which is a WCMCU -5102 module, based on PCM 5102a. Watching your diagram of connections I noticed that only DIN, BCK, LRC(LCK on my module) SCK(to Gnd) are connected. What about FLT, DMP, FMT, XMT ?… Are they going to ground, as I saw în a tutorial on the net ?!?…
Second. I have an IR RX but not TSOP 4838. It’s a VS 1838. With it’s own remote controller. Can I use it, or it’s mandatory to be an 4838 ?!?…
Thank you for your patience !😇😀

ovidiu169, 2023-11-21:
Hello Arne !
Thank you for your kind advice. I never could imagine that I could burn the DAC on 5 volts…
I will do as you mentioned and I will keep you posted !😀

Arne65, 2023-11-21:
Set SDMMC to 20000000 if you need a pullup then use it. I also had a similar problem. Not every card adapter always seems to work correctly, even if you use the card adapter on the display on a display it only works without a pullup on an identically constructed one with pullup. The card must be read correctly without additional buttons or anything similar.

ovidiu169, 2023-11-21:
Hello there !
The DAC PCM5102 is not working in any configuration…🫣
I am trying a VS1053 BoB v4 from Sparkfun. But what is “XS” on the diagram ? It is maybe CS on my board ?Also I don’t have GPIO 36 on Esp32 DevKit 1.,, Which the equivalent pin ?

schreibfaul1, 2023-11-21:
What kind of DAC do you have, this board? [image] Then check whether the bridges are inserted on the back.

GPIO36 is VN and GPIO39 is VP.
Are you sure that your ESP32 has a PSRAM?

ovidiu169, 2023-11-21:
Hi, Wolle !
No, I am not shure about this. In the beggining the serial monitor said “no PSRAM detected”. But now I’m not receiving this message again. Weird, huh ?!?…
Unfortunately I have another version of DAC. Is the same chip, but another configuration of PCB and pins. But is not working anyway. Maybe is defective from the very beggining. I ordered another one.
 Meanwhile I am trying to acomodate a VS 1053 BoB version from Sparkfun🫣
I never had a problem with periferals în my life. I am 63…😇

valera39, 2023-11-21:
Taking advantage of the fact that the author is present in the topic, I would like to ask: Mr. Wolle? Do you plan to please us with something new?

schreibfaul1, 2023-11-21:
Hi @valera39,
it's not quite clear yet what will come. I'm still waiting for the stable ArduinoV3.0/IDF5.1. from Espressif, which is expected to be released this year. New fonts (generated with LVGL Font Converter) are planned and already tested, which require far less flash and are written to the display faster. This means that larger characters are also possible, which are a prerequisite for 480x800px displays. The VS1053 HW decoder is no longer included. Only SW decoders that offer more options, such as progress bar, jumping forwards and backwards in files.... Possibly support for the KCX BT emitter. This is controlled via UART, can send and receive Bluetooth and is available at a reasonable price.

clear-sky, 2023-11-22:
hi dear wolle!
please let me better understand- your new plan is switch to such  displays?
https://www.buydisplay.com/7-inch-lcd-screen-tft-display-module-wvga-800x480-at070tn90-at070tn92

schreibfaul1, 2023-11-22:
Hi @clear-sky,
das ist zu viel Bastelei, das ist nur was für Experten, ich dachte an das:
https://www.elecrow.com/esp32-display-5-inch-hmi-display-rgb-tft-lcd-touch-screen-support-lvgl.html

ovidiu169, 2023-11-22:
Good morning everyone !!!
After two days of attempts to make working both modules(VS1053b and PCM 5102a) I have to surrender myself.
I tried everything possible but the result is zero, niente, nada !!!
It seems that I need to have the same modules as you used. Anything else is not working!!!
This is a paradox. Against of all the odds i succeeded to control the radio with a remote, but now I can’t hear anything…
[image]
[image]
[image]
[image]

[image]
[image]
[image]

ovidiu169, 2023-11-22:
Hello, Wolle !
Sorry for bothering you with my foolish questions… but do you think it is possible to have a Broken ESP board, meanwhile I had no result with any DACs i used ?!? 
At least one I am sure is working(VS1053b).
The other one is a weird version of PCM5102a(no, is not the version You showed before in your project). This one has a single row of pins. After I googled a lot it seems that all the pins(FLT, DMP, SCL, FMT,) are going to GND and XMT is going to 3.3V. But is not working anyway. I noticed a lot of people on the net had problems with this chip… Can I use an UDA ?!?

ovidiu169, 2023-11-22:
[image]

ovidiu169, 2023-11-22:
Finally, I made it !!! 
I would like to thank you, everyone who helped me.
All the best !!!😀

ovidiu169, 2023-11-23:
Hello there !
Is anyone who succeeded to add stations on csv. Stations file(Excell). I tried but nothing is happened. The radio plays the same stations!
I added two more(no.21 and 22) and i replaced the file on the sd card. No succes!
I do not know which is the cause. Also I do not imagine how else i could ediția this csv. File…
[image]
[image]

Dudinu, 2023-11-23:
Salut Ovidiu !
Am vazut ca se mai obisnuieste aici sa se raspunda si altfel decit in engleza :)
Am avut si eu o experienta asemenatoare cu editarea fisierelor .csv din acest proiect; initial am folosit Excel, se pare ca rezultatul nu este tolerat de proiect, asa ca acum folosesc SublimeText, cu care nu am avut probleme si care a venit instalat cu Visual Studio Code.
Sper sa fie de folos, toate bune !

ovidiu169, 2023-11-23:
Salut, Dinu !!!
Ce bine ca întâlnesc in sfârșit unul de-ai noștri pe aici !!!👍
Eu am probleme mari cu proiectul ăsta. Mulțumesc pentru pista, dar te rog frumos să fii mai explicit, nu sunt un as în programare🫣..,
Poți să-mi explici cum să fac, așa ca pentru un elev de clasa a cincea ?…
Please !!!

Jefke1, 2023-11-23:
Enter your IP address in the web browser. Then click on the search icon. Make your selection using the dropdown menus, use the arrows at the bottom to add the station and possibly station label. Fill in the edit popup and press OK. Press the second icon station list, click save.

schreibfaul1, 2023-11-23:
Exactly, you search for your station with the "Community Radio Browser". 
[image]

With (1) you can check whether the URL works. If no logo is supplied, you can enter the URL of the image in (2) via Google Image Search. The scaling is done automatically. If everything is OK, save the logo to the SD card with (3) and transfer the station data with (4) [image]
You will then find the new entry in the list (1). With (2) the list is transferred to the nvs of the ESP32

clear-sky, 2023-11-23:
[image]
hi! this is my dac-works fine

Dudinu, 2023-11-23:
Buna, Ovidiu !
Presupun ( din poza cu lista de canale ) ca lucrezi sub Windows; verifica in lista de programe ( Start -> Al Programs .....) daca ai instalat editorul "Sublime Text"; daca da, atunci conectezi cardul ( microSD, am vazut ca folosesti un adaptor) la PC, te duci pe fisierul Stations.csv , click dreapta pe mouse si in meniul care se deschide alegi Open With...   click normal. In lista care ar trebui sa se deschida cauti si alegi Sublime Text.
Solutia sugerata de Wolle mai sus ar trebui sa mearga DACA postul pe care vrei sa-l memorezi se gaseste in lista "Community Radio Browser".

clear-sky, 2023-11-23:
Hi dear wolle!
If you want your radio to be tested and used by many people around the world, it is better to make it so that it is modular so that parts can be obtained in different ways. It is definitely not everywhere like in Germany and many deprived countries do not have access to international stores.
Maybe it's better to use adapters like ft811 and fr810 to access a higher resolution, which also gives you a good hmi and only requires one spi port.
As you have seen, the parts can be found separately in the market. This module that you are considering cannot be found anywhere, in addition to other problems, for example, its sound is mono, also all the ESP32 pins are occupied, which makes possible development impossible.

ovidiu169, 2023-11-23:
Bună , Dinu !
Mulțumesc pentru răspuns, tu ești foarte amabil, vrei să mă ajuți, dar eu sunt prost grămadă . Am instalat SublimeText cu care am deschis fișierul Stations am adăugat posturi , am salvat, am înlocuit fișierul pe card, dar tot nu s-a întâmplat nimic.
Mai mult nu știu să fac.
Nici nu știu de unde scot băieții ăștia aplicația aia cu care poți să adaugi , să editezi, etc.., Tu știi cum ajungi la ea ?…
Din câte am văzut pe net, nu sunt singurul care nu înțelege . Și alții au tastat IP-ul în chrome și nu s-a întâmplat nimic… Nu apare nimic, nu ajungi nicăieri. Ori nu înțeleg eu, ori nu știu ei sa explice….

ovidiu169, 2023-11-23:
Hello, Clear-Sky !!!
PCM5102 wasn’t working even at 3.3v. 
The only DAC which was working for me was VS1053b , both versions(including Adafruit BoB)…

ovidiu169, 2023-11-23:
Dear friends !
First of all I want to say a Big Thank you for your patience with me.
Second, I am really puzzled.
I introduced my IP Adddress (that on the bottom side of the ILI display) on the Google Chrome but nothing is happening.
How do you reach the app named MiniWebRadio ?!?… I searched in the Community Radio Browser but I didn’t find it…
[image]
[image]
[image]

clear-sky, 2023-11-23:
clear all you see in address bar and type 192.168.0.213 and hit enter

ovidiu169, 2023-11-23:
Good evening You All,
Following the instructions You gave me I finally reach the MiniWebRadio App.
What I solved until now:
- acces the streams and settings;
- Select the stations;
- Checked if they are working properly;
- Checked the logos.
What I did NOT solved:
- saving the stations data on the card;
- Saving the logos also;
- The radio is not finding nothing on the list
- The app is losing connection every time I am trying to save data.
[image]
[image]
[image]
[image]

schreibfaul1, 2023-11-23:
I can't see the graphic for the buttons. The files are on the SD card in the png folder, is this missing?

ovidiu169, 2023-11-23:
The png files are on the content of the SD card... As a preview of this issue , I told you before I have no "buttons" also, on the display. So, the files are there, but...
[Content_on_SD_Card.zip](https://github.com/schreibfaul1/ESP32-MiniWebRadio/files/13453121/Content_on_SD_Card.zip)

ovidiu169, 2023-11-24:
Hello Wolle !
I just comment here to say the project is 90% ready.
I can control and play any station from my laptop, meanwhile I did not decoded the misteries regarding the addition of new stations on the SD card.
But taking into the consideration the precarious ESP32 board I had, for me is good enough.
I attached two pictures with my last web radios I’ve done. One is your concept and the other one is inspired by Mr. Volos(TTGO board).

Wish you all the best !!!
[image]
[image]

Arne65, 2023-11-26:
Connect to WiFi “MiniWebradio” under the specified IP. Then enter your SSID and password again. If that doesn't work, move closer to your router, your WiFi signal may be too weak.

ovidiu169, 2023-11-26:
Hallo Arne !
Simply, I can’t get it. I have the same router, with the same credentials(SSID and Password). 
Nothing was changed from previous version. My initial AP IP was another one which is working with my first version of radio. I introduced again(ssid/password) in common file and I saved it. I uploaded again on the board. Nothing !🙁
[image]
working on previous radio. How this 
[image]
was changed to 192.168.4.1(which is not opening) ?!?
The router is going, I can see the networks, but is not connecting to my MiniWebadio…
[image]
[image]

Arne65, 2023-11-26:
Is your PC connected to MiniWebradio accsespoint? Take a mobile device.
Otherwise I can't think of anything else. Can you ping 192.168.4.1 on your PC if you are connected to the Accsesspoint?

ovidiu169, 2023-11-26:
I tried. Nothing is happened. It's almost paranormal. That AP is not
responding at all.
I don't know what to believe. ... I don't know what to do...

schreibfaul1, 2023-11-26:
Two things to narrow down the error:
1) Does a simple WiFi example from Espressif work on the board?
2) The access data is written to the nvs. You can delete them with "erase flash", but first close the terminal, otherwise it won't work.

Arne65, 2023-11-26:
That's very surprising. You have to find the access point miniweb radio under the WiFi of your mobile device. If that's not the case, your ESP wouldn't send anything but receive nothing, which would then only indicate a defect in the WiFi area. Program a very simple one to test Wifi program on the esp and see if it works.

ovidiu169, 2023-11-26:
I tested the board, it seems to be defective !!!... It seems that it has
"unsupported crystal in use"(26Mhz)...
Chip is ESP32-DOWDQ6(revision 0).
Could you tell me which board You use to build this radio ?
I will try to find a similar one in order to check if and what I was doing
wrong....

ovidiu169, 2023-11-30:
Hello Wolle !I reached no result with my boards. Three of them(ESP32 DevKit 1). I closed the terminal and I erased flash of the devices. The result is the same. It is not connecting. Eventually the IP address is changed, but nothing more. It is clear for me it’s a connectivity problem. But how to solve it?!?…I documented myself on the net and I saw a lot of people having the same problem and nothing helped to solve it…Trimis de pe iPhone‑ul meuPe

zoli-iot, 2024-02-07:
> Hi @clear-sky, that's too much tinkering, it's only for experts, that's what I had in mind: https://www.elecrow.com/esp32-display-5-inch-hmi-display-rgb-tft-lcd-touch-screen-support-lvgl.html

Hi dear wolle!

Do you plan to support more convenient capacitive touch displays in the future alongside the traditional ILI9488?
For example:

https://www.aliexpress.com/item/1005006444056879.html

thanks for answer!

closed 2025-03-31 by github-actions[bot]