schreibfaul1/ESP32-MiniWebRadio #316: SD-Card Error when Bluetooth is connected

simusim, 2023-11-24:
Hi,

thanks for your fantastic radio!
But i found a strange problem.
I tried the Arduino V3 branch with A2DP-sink, really cool thing. Everything is working fine, but as soon as my phone is connected via Bluetooth the sd_mmc is throwing errors. Do you have any ideas about that? What is wrong here?

My setup is: 
-DollaTek ESP32 DevBoard, 4mb flash, 4mb psram.
-ILI9488 Display with internal cardreader. All resistors are bridged.

Here is my output from PlatformIO Monitor. Maybe someone could see something wrong?

<details><summary>Details</summary>
<p>

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13956
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
[   446][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

ESP32 Chip: ESP32-D0WD-V3
Arduino Version: 2.0.14
ESP-IDF Version: 4.4.5
MiniWebRadio     Version 2.18 Nov 12/2023
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 4194304 bytes, speed 40 MHz
CPU speed 240 MHz
SDMMC speed 40 MHz
TFT speed 40 MHz
PSRAM total size: 4192027 bytes

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V2     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 15219.9 MB by 15240.0 MB free
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  1754][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: 7-se
[  7699][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  7699][I][WiFiMulti.cpp:119] run(): [WIFI] 9 networks found
[  7700][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 2A:2D:ED:B4:88:4E SSID: 7-se Channel: 1 (-20)
[  9737][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
00:00:09 WiFI_info:   Connecting WiFi...
00:00:09 WiFI_info:   mDNS name: MiniWebRadio
00:00:09 setup: ....  connected to 7-se, IP address is 172.20.10.2
00:00:09 ftpServer:   Buffers allocated: 4622 bytes
00:00:09 rtime_info:  Initializing SNTP
00:00:09 rtime_info:  Waiting for system time to be set... (1/10)
12:42:18 rtime_info:  12:42:18
12:42:19 AUDIO_info:  commFMT = 1
12:42:19 audiotask:   is pinned to core 1
12:42:19 audiotask:   priority is 2
12:42:19 setup: ....  Number of saved stations: 14
12:42:19 setup: ....  current station number: 4
12:42:19 setup: ....  current volume: 3
12:42:19 setup: ....  last connected host: https://extra-radio.radionetz.de/extra-radio.mp3
12:42:19 setup: ....  connection timeout: 500 ms
12:42:19 setup: ....  connection timeout SSL: 2000 ms
12:42:19 action: ...  current volume is 3
12:42:19 action: ...  switch to station 4
12:42:19 AUDIO_info:  Connect to new host: "https://www.extra-radio.de/stream/listen.m3u"
12:42:19 AUDIO_info:  PSRAM found, inputBufferSize: 638965 bytes
12:42:19 AUDIO_info:  buffers freed, free Heap: 40344 bytes
12:42:21 AUDIO_info:  SSL has been established in 1948 ms, free Heap: 30164 bytes
12:42:21 AUDIO_info:  Connect to new host: "https://extra-radio.radionetz.de/extra-radio.mp3"
12:42:21 AUDIO_info:  buffers freed, free Heap: 39556 bytes
12:42:24 AUDIO_info:  SSL has been established in 2169 ms, free Heap: 33376 bytes
12:42:24 WiFI_info:   RSSI is -23 dB
12:42:24 DLNA_info:   SSDP search for media servers started
12:42:24 bitRate:     128Kbit/s
12:42:24 bitRate:     128Kbit/s
12:42:24 icy-descr:   Mein Hochfrankenradio
12:42:24 AUDIO_info:  icy-name: extra-radio
12:42:24 StationName: extra-radio
12:42:24 icy-url: ..  http://www.extra-radio.de
12:42:24 AUDIO_info:  MP3Decoder has been initialized, free Heap: 17232 bytes , free stack 2252 DWORDs
12:42:24 lastURL: ..  https://extra-radio.radionetz.de/extra-radio.mp3
12:42:25 AUDIO_info:  stream ready
12:42:25 AUDIO_info:  syncword found at pos 0
12:42:25 AUDIO_info:  Channels: 2
12:42:25 AUDIO_info:  SampleRate: 44100
12:42:25 AUDIO_info:  BitsPerSample: 16
12:42:25 AUDIO_info:  BitRate: 128000
12:42:25 StreamTitle: Howard Carpendale - Laura Jane 
12:42:28 DLNA_info:   SSDP query discovered 0 media servers
12:42:28 ftpServer:   Ftp server waiting for connection on port 21
12:42:29 Audiocodec:  MP3
12:42:39 BT state:    BT enabled
12:42:49 BT state:    A2DP connection state: Connecting
12:42:50 BT info:     SampleRate: 44100 Hz
12:42:50 BT info:     remote rn_cap: count 7, bitmask 0x1f06
12:42:50 BT state:    A2DP connection state: Connected
E (43164) diskio_sdmmc: sdmmc_read_blocks failed (257)
E (43165) diskio_sdmmc: sdmmc_read_blocks failed (257)
12:42:50 AUDIO_info:  file "/common/m/BT.jpg" not found
E (43179) diskio_sdmmc: sdmmc_read_blocks failed (257)
E (43180) diskio_sdmmc: sdmmc_read_blocks failed (257)
12:42:50 AUDIO_info:  file "/common/m/unknown.jpg" not found
E (43888) diskio_sdmmc: sdmmc_read_blocks failed (257)
E (43889) diskio_sdmmc: sdmmc_read_blocks failed (257)
12:42:51 AUDIO_info:  file "/common/m/RSSI4_bt.bmp" not found
Disconnected (GetOverlappedResult failed (PermissionError(13, 'Zugriff verweigert', None, 5)))

</p>
</details> 

I would be happy if someone could help me :)

Arne65, 2023-11-24:
Hello , 

Have you updated the data on your SD card with the SD.Zip from the Arduino 3.0 branch. They will be missing the icons

simusim, 2023-11-24:
Yes, I did.
The files are on the sd card and sometimes (really sporadic) they are displayed. 
I'm confused about the "sdmmc_read_blocks failed (257)"-error, it looks like the esp has no access to the card in this moments?

Arne65, 2023-11-24:
I think you are using the SD slot of the display. Despite bridging the resistors, I also had small problems with it. A pullup of 10k ((D0) solved that for me

simusim, 2023-11-25:
Thanks for your hint, but sadly that did not change something.

In normal radio mode the sdcard works without problems. 
So could it probably be an power-issue with the onboard 3V3-regulator? Maybe it is not able to source enough current, so if bluetooth is powerhungry the voltage drops/break?
But I have no idea how to test..

Arne65, 2023-11-25:
The collapse of the tension would be a plausible explanation for your problem. What frequency for the SD card are you working with? If you work with 40MHz or higher, reduce the frequency to 20MHz.

Jefke1, 2023-11-27:
[image]
I just built V3. External SD reader on 20Mhz. Esp32 Wrover IE.
Just usb powerd from my laptop. 
1 small problem, the V3 start-up screen is only half loaded, but after that everything works like a charm. 
Maybe not enough psram, but I'm not sure about this, because I don't have a 4mb board to test it.

schreibfaul1, 2023-11-27:
The V3 branch is in the testing phase, there are still some bugs to be fixed before it becomes master.
@simusim if the SD card works otherwise, I don't think it's a problem with the operating voltage. The error appears exactly when the BT logo changes from silver to gold. I suspect a reading error of the file: /common/s/BT.jpg or /common/m/BT.jpg, (depending on the display size).
@Jefke1 probably a bug related to the jpg decoder. I have created some images with 16x16px, but the decoder prefers 8x8px blocks. I'll take a look at that.

Arne65, 2023-11-27:
Hello,
I have the Arduino 3.0 branch on a Wrover VIE 8MB Flash 8MB PSRAM and it works perfectly.