schreibfaul1/ESP32-MiniWebRadio #318: audiotask.cpp compiling error

TubaFonse, 2023-12-01:
Dear Wolle, I need your help: 
I get the following compiling error, but only with your actual Firmware (in ESP32-S3-DevKitC-1-N16R8 / PCM5102A configuration): 

Building in release mode
Compiling .pio\build\esp32s3\src\audiotask.o
Compiling .pio\build\esp32s3\src\main.o
src/audiotask.cpp: In function 'void audioTask(void*)':
src/audiotask.cpp:359:62: error: no matching function for call to 'Audio::setPinout(int, int, int, int, int)'
     audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, -1, I2S_MCLK);
                                                              ^
In file included from src/audiotask.cpp:15:
.pio/libdeps/esp32s3/ESP32-audioI2S-master/src/Audio.h:146:10: note: candidate: 'bool Audio::setPinout(uint8_t, uint8_t, uint8_t, int8_t)'
     bool setPinout(uint8_t BCLK, uint8_t LRC, uint8_t DOUT, int8_t MCK = I2S_GPIO_UNUSED);
          ^~~~~~~~~
.pio/libdeps/esp32s3/ESP32-audioI2S-master/src/Audio.h:146:10: note:   candidate expects 4 arguments, 5 provided
*** [.pio\build\esp32s3\src\audiotask.o] Error 1

I tried to change audiotask.cpp (line 359) from 
audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, -1, I2S_MCLK); 
to (4 arguments)
audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, I2S_MCLK); 

the compiler finished without error. 
After flashing that new Firmware my MiniWebRadio plays, but Touchpanel didn't work (no reaction). 
So I actually flashed the previous Fw-Release (Sep 2023) to get back my MiniWebRadio without problems. 

many thanks, 

TubaFonse

schreibfaul1, 2023-12-01:
Thanks for the bug report. Today I upgraded the audio lib from 2.0.14 to Arduino V3.0.0. And forgot to adjust the audiotask.

TubaFonse, 2023-12-02:
Thanks - solved - but Touchpanel didn't work (no reaction). 
I'll open a new issue for that problem with details.

closed 2023-12-02 by TubaFonse