schreibfaul1/ESP32-MiniWebRadio #319: Nothing happens after Logo appears on the LCD

JozefKc, 2023-12-02:
After reset of CPU logo appears on the LCD /ILI9341/, on the monitor there is:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
[   956][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[  1599][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: -3-
[  1611][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: -1-
[  1611][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: -3-
[  8302][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  8302][I][WiFiMulti.cpp:119] run(): [WIFI] 15 networks found
[  8304][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: F0:79:59:D1:25:4C SSID: -3- Channel: 1 (-34)
[ 11609][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.

White WLED on the CPU PCB goes on and ..... nothing - see the video.

Only in 1 of 100 attempts radio starts.  Then Touch works, connection per FileZilla and everything is OK. I can not find the reason..... CPU is ESP32S3-WROOM1-N16R8, PSRAM changed to octal......
Have you any advice plaease?

https://github.com/schreibfaul1/ESP32-MiniWebRadio/assets/120331511/17bf97c8-dcc8-4ac6-9536-05603c4f3762

[image]

TubaFonse, 2023-12-02:
I use the same configuration (ESP32-S3-DevKitC-1-N16R8 CPU with ILI9341 Display, PCM5102A Decoder). 
You know, there are many pin-restrictions with that CPU (16MB Flash, 8MB octo PSRAM)? 
I'm using 
I/O 6 (instead of I/O 39) for TP_IRQ 
I/O 5 (instead of I/O 3) for I2S_BCLK 
I set I2S_MCLK to -1 (instead of I/O 0, because PCM5102A don't use I2S_MCLK) 

in that configuration my radio plays, but now only with actual firmware version my touchpanel didn't work.

minicico, 2023-12-10:
I have the same problem. Only the introductory logo starts and according to the output after serial communication it stops at the initialization of the VS1053 module. Once in x times it starts up fine.

JozefKc, 2023-12-11:
Finaly, after the investigation of the reason of freezing the system I found that there was a shorcut between MISO and nearby pin on the processor. This is the frequently error..... Try to meassure the resist between nearby pins. Of course, processor must be out of the PCB.
For improving of the start I do not use 3,3V from the processor board for supply of the display. I installed a separate AMS1117 for 3,3V for display Vcc.

RobinMosedale, 2024-02-17:
Having now followed the advice, I'm past the boot stage of reading the SD, 1k pullup.

However, I'm now encountering exactly similar symptoms, stuck on the welcome page. I suspect that it's the VS1053 initialisation, checked the wiring, and may substitute a spare VS1053. However, I'm guessing.
I'm not getting any serial terminal output, and somewhat puzzled how I switch that on. I've tried setting in platformio.ini
-DCORE_DEBUG_LEVEL=3     to either 4 or 5, but nothing.
Could anyone please advise how I switch on serial monitoring of any programme messages, indeed if there are?
[image]

RobinMosedale, 2024-02-17:
My apologies, just done a clean load and getting serial output and that's confirming Vs1053 failure
[image]

closed 2025-03-01 by github-actions[bot]