schreibfaul1/ESP32-MiniWebRadio #321: Arduino V3.0 sometimes no station information

Arne65, 2023-12-13:
Hallo Wolle ,

erstmal vielen Dank für Deine ausgezeichnete Arbeit.

Ich habe mal den neuen Arduino V3.0 Branch auf eine ESP32 WROVER N8R8 getestet soweit funktioniert alles ohne Probleme. 
Bei einigen Sendern gibt es aber keine Titel Infos mehr auf dem Display bei anderen dann wiederum werden diese angezeigt . Das gleiche gilt dann auch für den seriellen Monitor.

In der ersten Testversion fdfa4bf0 war das alles ok.

schreibfaul1, 2023-12-14:
Hi Arne,
das kann ich nachvollziehen. Ursache ist die neue DLNA Bibliothek. Ich werde noch eine Weile daran bauen, damit DLNA vom Display bedient werden kann. Es gibt dann wahrscheinlich am Wochenende ein Update. Im V3 Branch sind noch weitere Fehler, so schnell wird der nicht zum Master.
beste Grüße
Wolle

Arne65, 2023-12-22:
Hallo Wolle ,

was meinst du mit (RT) am BT-Emitter ?

Schöne Weihnachtsfeiertage und 

Beste Grüße

schreibfaul1, 2023-12-22:
Hallo Arne,
der V3 Branch ist noch in der Entwicklung. Da sind noch viele Fehler drin.
Der KCX BT Emitter soll die Verbindung zu externen BT Geräten herstellen, speziell zu Kopfhörer und Lautsprecher mit BT Funktion. BT und WiFi geht auch beim ESP32 nicht. Auch mit künftigen ESP µC wird das nicht möglich sein. Diese Lücke soll der BT Emitter schließen, zumal dieser kostengünstig und bei den großen Plattformen überall erhältlich ist. Mit der Größe einer Briefmarke ist das Löten an den Pins nicht ganz einfach. Ich habe ihn auf eine Adapterplatine gelötet, RT ist der Anschluss mit dem Draht ganz rechts. 
[image]
 RT schaltet zwischen Senden/Empfangen um. Standardmäßig ist RT HIGH, die rote LED leuchtet und der Sender ist aktiv, wird RT LOW und ein SW RESET ausgeführt, dann leuchtet die blaue LED und der Empfänger ist aktiv.
[image]

closed 2025-03-01 by github-actions[bot]