schreibfaul1/ESP32-MiniWebRadio #322: Taster einbinden

fv1895, 2023-12-15:
Hallo zusammen,

ich würde gerne 3 Taster in das Programm integrieren, scheitere aber kläglich:
1. nextstation
2. prevstation
3. mute

Bei meinem Versuche ist der Taster quasi immer gedrückt, was mache ich falsch?

Ich habe es wie folgt versucht:
`void setup() {
    Serial.begin(115200);
    Serial.print("\n\n");

    pinMode(35, INPUT_PULLUP);`
`void loop() {
    if(!_f_ESPfound) return;    // Guard:  wrong chip?
    if(!_f_SD_MMCfound) return; // Guard:  SD_MMC could not be initialisized
    webSrv.loop();
    ir.loop();
    tp.loop();
    ftpSrv.handleFTP();
    soap.loop();

 if (digitalRead(35) == HIGH) {
 SerialPrintfln ("Taster ein");
 }`

Ich würde mich sehr über eine Lösung freuen!
Schönen Gruß

Marcus

schreibfaul1, 2023-12-15:
Hallo Marcus,
mechanische Tasten neigen zum Prellen. Daher ist es besser eine Wartezeit einzubauen.
Bsp. (nicht getestet, wird aber funktionieren)
`````c++
void loop(){
...
    static bool bt = false;
    static uint32_t ms = 0;
    if(digitalRead(35) == LOW && bt == false){
        bt = true;
        nextStation();
        ms = millis() + 200; // wait 200ms
    }
   if(bt == true && ms < millis()) bt = false;
....
}
````
Der ESP32 hat an PINs 34...39 keine internen PullUps,  du müsstest dann einen von außen (~1...50K) anbauen

fv1895, 2023-12-15:
Vielen Dank!
Ich werde es ausprobieren.

fv1895, 2023-12-18:
Passt jetzt nicht zu dem Thema aber jetzt ist ein weiteres Problem aufgetaucht, der Touchscreen reagiert nicht richtig. Ich bekomme immer folgende Fehlermeldung:
Touchpoint not valid

Am TFT liegt es nicht, habe das selbe Problem auch auf einem neuen Display.
Woran kann das liegen?

closed 2025-03-01 by github-actions[bot]