schreibfaul1/ESP32-MiniWebRadio #323: KCX_BT_Emitter for V3

ve2cfb, 2023-12-23:
Hello Wolle,
for those interested in a BreakOut Board for that BT Module, a fellow by the name of Ralph Bacon, did one.
you can check this github Web Site...  but could not find how to order it.
https://github.com/RalphBacon/235_KCX_BT_EMITTER-Breakout-Board  

Also, i do have different version of that KCX_BT module and was wondering if using the AT command, i could find the BT version.
Looking at your BT module, it seems to be at V5.1 or v5.3  (search Kcx_BT_emitter in Aliexpress.com)
It seems that both board have v1.7 but have different chip size and depending on the chip size it could be V4.1 or v5.1 or v5.3

Pierre

schreibfaul1, 2023-12-23:
Hello Pierre,
thank you for your contribution. I have the board with v1.7. Internally, this board has a memory for 10 BT MAC addresses. So you can define in a table which devices should pair. The connection is re-established super-fast with the last known device. And it can send or receive. Disadvantage: it cannot process metadata (?). Other modules (from Aliexpress) may be similar or perhaps even better. Unfortunately I don't know, I don't have the knowledge and experience, but I would like to add BT functionality. And it must not be a module that is difficult or expensive to obtain.

ve2cfb, 2023-12-24:
From my pass experience with those chinese BT EMITTER, i had mixed success with V4.1. They work with some bt speaker but not on others.
On aliexpress, they sell a new KXC_BT_EMITTER at version 5.1 and 5.3 with the same chip size as your and mine.
On the back of those module they all have V1.7 go figure...

I do have a few KCX_BT_EMITER with different chip size and i will try to get the BT Version of those chip.
https://github.com/Mark-MDO47/BluetoothAudioTransmitter_KCX_BT_EMITTER 
In the code file there is a section that will surely do that..

// < 3>: query software version
// Description: Check the software version; my devices show V1.1         ( *** that is probably the back pcb version)
//   AT+GMR?
//     OK+
//     KCX_BTEMITTER_Vx.x

schreibfaul1, 2023-12-25:
Then I have a very old board.
AT+GMR? responds with OK+VERS:KCX_BT_RTX_V1.4
In TX mode, my BT headphones are recognised, but the Android smartphone is not. In RX mode, I can connect to the smartphone.

ve2cfb, 2023-12-26:
Same thing here. I got V1.4
So we still have a problem. Is it the board version or the BT software version.
All my pcb board are V1.7 
chip id: AC22b9of671  impossible to communicate. Size of chip: 4x8mm
chip id: BPoW17S         i was able to communicate with it.  Size of chip: 4x4mm

I will order a couple of those new one and see they send me the right version. (apparently at V5.x)

https://www.aliexpress.com/item/1005005432551133.html?spm=a2g0o.productlist.main.19.12857cdb5bl6ku&algo_pvid=658176e3-effc-447b-b849-f3df1994ff8c&algo_exp_id=658176e3-effc-447b-b849-f3df1994ff8c-9&pdp_npi=4%40dis%21CAD%214.18%212.59%21%21%213.08%21%21%40210313e917035964094618471ef4dc%2112000033053244147%21sea%21CA%21773063114%21&curPageLogUid=H7XtVOtLHJBf#nav-review 

https://www.aliexpress.com/item/1005005550299725.html?spm=a2g0o.productlist.main.11.12857cdb5bl6ku&algo_pvid=658176e3-effc-447b-b849-f3df1994ff8c&algo_exp_id=658176e3-effc-447b-b849-f3df1994ff8c-
5&pdp_npi=4%40dis%21CAD%212.75%211.31%21%21%212.03%21%21%40210313e917035964094618471ef4dc%2112000033708007253%21sea%21CA%21773063114%21&curPageLogUid=ZmCS7xOqvwgx

schreibfaul1, 2023-12-27:
My chip is square and has the inscription ''BP0U598-56C8"

ve2cfb, 2024-01-07:
Hello Wolle,
Quick Question...
on your BT KCX_BT_EMITTER, what does RT refer too ?
does it refer to DP or DM

Many thanks again.  
Pierre

Arne65, 2024-01-07:
Hello Piere,

[image]

ve2cfb, 2024-01-07:
Helleo Arne65.
I see where my problem is... I do have the following module type v1.2 and v1.4
Last week, i did order a couple of module like yours.
Thanks for the info.

[image]
[image]

[image]

schreibfaul1, 2024-01-07:
The RT pin is used to switch between transmitting and receiving. I am not quite finished with the integration of the KCX_BT_Emitter and the RT pin currently has no function.

ve2cfb, 2024-01-07:
This page is quite usefull: especially about the mode jumper. (See page 3  of that Web Page and do a search on **"Bottom"**)
Very good Info.

https://www.electro-tech-online.com/threads/kcx_bt_emitter-low-cost-bluetooth-bt-audio-module.158156/

JozefKc, 2024-04-16:
Hi Wolle!
Does it mean that it is not necessary to connect MODE (Tx/Rx) to ESP?  I am just going to add the BT 1.7 to the radio.
Thanks in advance.

schreibfaul1, 2024-04-17:
If the MODE connection remains free, the transmit mode will always be active. This is sufficient for operating an external BT loudspeaker or voice assistant such as Alexa.

JozefKc, 2024-04-25:
Does it also mean that RT wire from ESP shall be connected to the RX port on the KCX_BT card? See picture....
[image]

schreibfaul1, 2024-04-25:
Exactly, the pin switches between transmitting/receiving. This is the "MODE" pin. 
I have changed the designation, MODE and RT are the same.

valera39, 2024-06-13:
Hi Dear Wolle!
Very often the radio playback is interrupted and such records appear in the log:
23:23:44 BT-Emitter:  Status -> Disconnected
23:23:45 BT-Emitter:  Status -> Scan...
23:23:56 BT-Emitter:  POWER ON 
23:23:56 BT-Emitter:  Status -> Disconnected
23:24:03 WiFI_info:   RSSI is -56 dB
23:24:04 BT-Emitter:  POWER ON 
23:24:04 WiFI_info:   RSSI is -49 dB
23:24:05 BT-Emitter:  Status -> Disconnected
23:24:06 BT-Emitter:  CON ONE 
23:24:06 BT-Emitter:  CONNECT=>MacAdd:9c1fbeabc930,Name:KDL-43WF804
23:24:27 BT-Emitter:  POWER ON 
23:24:27 WiFI_info:   RSSI is -47 dB
23:24:28 BT-Emitter:  Status -> Disconnected
23:24:28 BT-Emitter:  CON ONE 
23:24:28 BT-Emitter:  CONNECT=>MacAdd:9c1fbeabc930,Name:KDL-43WF804
in this case my bluetooth module is connected to my neighbor's TV.
How can I get rid of the discontinuities?
Also sometimes when I turn on the radio, the bluetooth module switches to receiver mode before the station is loaded, I have to switch it manually to emitter mode to get sound

schreibfaul1, 2024-06-15:
Yes, that's right, there is an "Auto connect" that connects as soon as a BT device is recognised. To prevent this, you must enter your device or devices in the table. Every device found during scanning is suggested, if you know the BT device name or MAC address you can also enter it manually. As soon as there is an entry in the table, "Auto connect" is deactivated. The BT Emitter can store up to 10 devices.
[image]

valera39, 2024-06-15:
So auto-connect is only enabled when a device that is not on the list appears, correct?

schreibfaul1, 2024-06-15:
Exactly, if there is an entry, it will only connect to the devices in the list. You cannot delete a single entry from the list individually. Only the entire list with the red "Clear" button or you can press and hold the button on the KCX-BT_Emitter.

Arielhh, 2025-04-16:
Hey, has anyone run into Wi-Fi and Bluetooth interfering with each other ? how can it be solved?

schreibfaul1, 2025-12-16:
Hello @kaernter53,
I don't have a circuit diagram for the board, but it looks as if the capacitors are located at the audio input. That could be a low-pass filter. Thanks for pointing that out.