schreibfaul1/ESP32-MiniWebRadio #325: VS1053 support removed

Rhombur, 2023-12-26:
Is the VS1053 support removed? I tried to build today, and it failed with the messages below. The audio decoder was set to 0 = VS1053. The new audiotask (V3.0.0) does not contain anything with respect to VS1053 as previously. Is this intended?
Many thanks

```
src/audiotask.cpp: In function 'void audioTask(void*)':
src/audiotask.cpp:47:21: error: 'I2S_BCLK' was not declared in this scope
     audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, I2S_MCLK);
                     ^~~~~~~~
src/audiotask.cpp:47:21: note: suggested alternative: 'I2C_CLK'
     audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, I2S_MCLK);
                     ^~~~~~~~
                     I2C_CLK
src/audiotask.cpp:47:31: error: 'I2S_LRC' was not declared in this scope
     audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, I2S_MCLK);
                               ^~~~~~~
src/audiotask.cpp:47:40: error: 'I2S_DOUT' was not declared in this scope
     audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, I2S_MCLK);
                                        ^~~~~~~~
src/audiotask.cpp:47:40: note: suggested alternative: 'FM_DOUT'
     audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, I2S_MCLK);
                                        ^~~~~~~~
                                        FM_DOUT
src/audiotask.cpp:47:50: error: 'I2S_MCLK' was not declared in this scope
     audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, I2S_MCLK);
                                                  ^~~~~~~~
src/audiotask.cpp:47:50: note: suggested alternative: 'I2C_CLK'
     audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, I2S_MCLK);
                                                  ^~~~~~~~
                                                  I2C_CLK
*** [.pio\build\esp32s3\src\audiotask.o] Error 1
```

Arne65, 2023-12-26:
Hello Rhombur,

The Arduino Branch 3.0 only works with a DAC, the VS 1053 cannot be used

Rhombur, 2023-12-26:
I did not use the Arduino-V3.0 branch, but the default master.

Arne65, 2023-12-26:
As far as I know, this also applies to the master's degree

Rhombur, 2023-12-26:
Indeed, this is what it looks like - as the description, how to install etc. all tell a VS1053 can be used (which was true at least until few weeks ago). Seems no chance to use my new ESP32-S3 whichw as supposed to solve the performance issues over my older EPS32.

Arne65, 2023-12-26:
You would have to switch from the VS 1038 to a DAC then it will work

Arne65, 2023-12-26:
When using the VS 1083, Bluetooth cannot be used; it only works with a DAC

Rhombur, 2023-12-26:
I have VS1053 at hand, but I don't have DAC ;)  Would need to order it first, and wait again. I don't care for Bluetooth.

Arne65, 2023-12-26:
If I understood all of this correctly, the VS 1083 no longer offers any further developments in the radio. I think that's probably why Wolle deactivated the VS1053 ,but I'm not entirely sure. I only use DAC myself

Arne65, 2023-12-26:
If I understood all of this correctly, the VS 1053 no longer offers any further developments in the radio. I think that's probably why Wolle deactivated the VS1053,but I'm not entirely sure. I only use DAC myself

Rhombur, 2023-12-26:
Seems I have to move this path as well. For now, need to find a way to work with the V2.14 from local drive.

Linkenelis, 2023-12-31:
Bummer, I feel a VS 1053 sounds better. Is that just a feeling? I am still using older version. I Like the DLNA development, so....