schreibfaul1/ESP32-MiniWebRadio #327: 2. Hochladen auf ESP32 schlägt immer fehl

fetzer91, 2024-01-03:
Hallo schreibfaul,
ich habe den Code kompiliert bekommen und scheitere nun aber am Upload. er versucht 2 Uploads hintereinander von denen der zweite (esp32s3) immer fehlschlägt. Ich habe es schon mit der Bootknopf und auch mit der Kondensator-variante versucht aber nichts half. Was könnte noch das Problem sein? ich habe hier zwei ESP´s liegen und bei beiden tritt das Problem auf.
Hoffe auf Erleuchtung..

VG Stefan

schreibfaul1, 2024-01-03:
guck mal bitte unten in der Statuszeile, ob du die richtige Umgebung ausgewählt hast
[image]

AVRsteffen, 2024-01-03:
Toll, wo findet man den Schalter dafür?

schreibfaul1, 2024-01-03:
Einfach auf den Eintrag mit der linken Maustaste klicken, dann öffnet sich ein Auswahlmenü. Wenn keins kommt gibt es wahrscheinlich ein Problem mit VS Code (dann bleibt meistens nur Neuinstallation)
[image]

fetzer91, 2024-01-03:
Aha...das kannte ich noch nicht. Hat jetzt mit dem zweiten Eintrag env:esp32 funktioniert.
Viiiiiiielen Dank für deine schnelle Hilfe!

Kurze andere Frage noch.. Sehe ich den ESP nach dem flashen ohne die restliche Verkabelung schon im Netzwerk?

AVRsteffen, 2024-01-03:
Thanks Wolle 👍

schreibfaul1, 2024-01-03:
Die SD_MMC muss auf jeden Fall angeschlossen und auf der SD Karte muss die zip Datei entpackt werden.
Offene Eingänge sind immer ein Problem, aber grundsätzlich wird dann der AccessPoint geöffnet und du kannst die WLAN Zugangsdaten über ein Smartphone eintragen. Wenn dein Router dann eine IP vergeben hat, siehst du auch die Webseite.

fetzer91, 2024-01-03:
Okay, dann mach ich mir mal keine Sorgen. Hab vielen Dank für deinen Support!
VG Stefan