schreibfaul1/ESP32-MiniWebRadio #328: VS1053

maba64, 2024-01-08:
Hallo
wenn ich den VS1053 aktiviere erhalte ich immer Fehlermeldung (platformio)
Hast du mir da ev einen Tipp ?
besten Dank
Markus

schreibfaul1, 2024-01-08:
Hallo,
um den Fehler einzugrenzen kannst du alles in main.cpp löschen und durch den Beispielcode von hier https://github.com/schreibfaul1/ESP32-vs1053_ext ersetzen (und die  WiFi Zugangsdaten und GPIOs anpassen)
Dann gestaltet sich die Fehlersuche einfacher.

maba64, 2024-01-08:
besten dank schau mir dies gerne an

noch eine Frage
betreffend dem Gehäuse, kanns Du mir ev das Druckfile senden?
besten Dank
Markus

Arne65, 2024-01-09:
Hallo Markus 

Ich habe das so gelöst . Vielleicht kann es dir ja helfen .
[image]
[image]

closed 2024-02-24 by github-actions[bot]