schreibfaul1/ESP32-MiniWebRadio #329: Hang when no audio file in AudiFiles  base directory

ve2cfb, 2024-01-08:
Hello Wolle,

did a search in ISSUES but nothing found.
my audiofiles directory is as follow...
audiofiles......
                    Pop
                    Medley
                    Rieu
                    Chrismas
test.mp3

1- if i remove the test.mp3,  the system hang and after a minute or so, does a reset.
2- copy the test.mp3 back to audiofiles base directory and everything is ok, no more problem. 

---- Opened the serial port COM8 ----
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400dc5e0  PS      : 0x00060830  A0      : 0x800def7e  A1      : 0x3ffdbea0  
A2      : 0x00000006  A3      : 0x00000000  A4      : 0x3ffc758d  A5      : 0x3ffc750c  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800dc5d0  A9      : 0x3ffdbe40  
A10     : 0x1439e318  A11     : 0x3ffc9acc  A12     : 0x1439e318  A13     : 0x3f4d1159  
A14     : 0x3f401a13  A15     : 0x3ffc50bc  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4009268a  LEND    : 0x40092695  LCOUNT  : 0x00000000  

Backtrace: 0x400dc5dd:0x3ffdbea0 0x400def7b:0x3ffdbf20 0x4011b122:0x3ffdc040 0x400e0cb3:0x3ffdc060 0x40169014:0x3ffdc0b0

ELF file SHA256: 852e30935a942c45

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13956
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
[   446][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

ESP32 Chip: ESP32-D0WD-V3
Arduino Version: 2.0.14
ESP-IDF Version: 4.4.5
MiniWebRadio     Version 3.00f Jan 04/2024 
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 8388608 bytes, speed 40 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 4191995 bytes

13:30:19    
13:30:19        ***************************    
13:30:19        *     MiniWebRadio V3     *    
13:30:19        ***************************    
13:30:19    
13:30:19 setup: ....  Arduino is pinned to core 1
13:30:19 tft_info: .  init ILI9488
13:30:19 setup: ....  Init SD card
13:30:19 setup: ....  SD card found, 5779.8 MB by 15001.5 MB free
13:30:20 setup: ....  seek for stations.csv
13:30:20 setup: ....  stations.csv found
13:30:20 setup: ....  seek for WiFi networks

The rest is as usual.

Pierre

schreibfaul1, 2024-01-08:
Hello Pierre,
I can't reproduce thist. There is a memory corruption, unfortunately without any further information. I assume you deleted the last audio file via the website. This works for me without any errors.

Can you please rename the existing folder "audiofiles" and create a new folder. If the error then disappears, there is a problem with the SD card. If the serial terminal is active, are there any other hints?

ve2cfb, 2024-01-09:
Hello Wolle,
i did what you asked and i get same result.
i change the ESP32-S3 for a ESP32  8 Mbytes and same problem.
It is not a big deal. I copied a mp3 song in audiofiles and its ok.

closed 2024-02-24 by github-actions[bot]