schreibfaul1/ESP32-MiniWebRadio #330: RGB LED on ESP32-S3 N16R8  and Bluetooth KCX-BT

ve2cfb, 2024-01-09:
Hello Wolle,
On my ESP32-S3 N16R8 WROOM (red pcb board) , my RGB Led pin 48 is alway on and looking at your BT pin config, the #define BT_EMITTER_RX is  pin 48. I tried a different gpio45 and i got rid of the RGB light (it is very bright). 
Since you are the big BOSS, can you suggest different GPIO for  BT_EMITTER_RX and BT_EMITTER_TX.

On a side note, it looks like you will be using more AT command with the RX/TX pins and  on my previous version of the KCX_BT i had the CON pin of that BT module going to a push button. Maybe i dont need that switch anymore with your V3. 

This project is getting better and better.
Thanks very much for your help.
Pierre

JozefKc, 2024-01-13:
I have the same problem. The LED is very annoying.

schreibfaul1, 2024-01-13:
I have changed this, the new RX pin is 45. Pin 48 seems to be the RGB LED on many boards. So the RGB LED can still be used for other purposes.

AVRsteffen, 2024-01-13:
The esp32 is slowly running out of pins... couldn't you make an extension like spi and 74lvc595?  Didn't even the SPI of the VS1053 become free again?  Slow outputs such as CS, C/D or AMP on/off, mute etc.  could then be taken over by the 595. The two pins for CS TFT and touch could be operated via a single pin + inverter.

schreibfaul1, 2024-01-14:
You will no longer find the VS1053 in future versions. Only external DACs will be supported. If you replace the VS1053 with a PC5102, you will get free pins. This is certainly better than a port expander or shift register. An ESP32-S3 provides further free pins.

JozefKc, 2024-01-18:
By exchanging VS1053 to PCM5102 RGB problem disapeared. Also disapeared freezing of the radio at the start. I do propose to leave VS1053. With PCM5102 also the VU meter started to work! With VS1053 there was also the problem to play .ogg format. Now it works.

closed 2024-03-04 by github-actions[bot]