schreibfaul1/ESP32-MiniWebRadio #331: Compiling Errors in audiotask.ccp

OldTinkerer, 2024-01-09:
Hi Radio-Listeners,

today I tried -according the installing description- to get the WEB Radio V2 running, but some errors...

SETUP:
ESP32-S3-WROOM1   N16R8 (I did all necessary changes in source and PlatformIO IDE)
ILI9488
VS1053B

Errors in Terminal (I cutted the previout part, because it was OK):

Compiling .pio\build\esp32s3\cbor\tinycbor\src\cborparser_dup_string.o
Compiling .pio\build\esp32s3\cbor\tinycbor\src\cborparser.o
Archiving .pio\build\esp32s3\esp-idf\bootloader_support\libbootloader_support.a
src/audiotask.cpp: In function 'void audioTask(void*)':
src/audiotask.cpp:47:21: error: 'I2S_BCLK' was not declared in this scope
     audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, I2S_MCLK);
                     ^~~~~~~~
src/audiotask.cpp:47:21: note: suggested alternative: 'I2C_CLK'
     audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, I2S_MCLK);
                     ^~~~~~~~
                     I2C_CLK
src/audiotask.cpp:47:31: error: 'I2S_LRC' was not declared in this scope
     audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, I2S_MCLK);
                               ^~~~~~~
src/audiotask.cpp:47:40: error: 'I2S_DOUT' was not declared in this scope
     audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, I2S_MCLK);
                                        ^~~~~~~~
src/audiotask.cpp:47:40: note: suggested alternative: 'FM_DOUT'
     audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, I2S_MCLK);
                                        ^~~~~~~~
                                        FM_DOUT
src/audiotask.cpp:47:50: error: 'I2S_MCLK' was not declared in this scope
     audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, I2S_MCLK);
                                                  ^~~~~~~~
src/audiotask.cpp:47:50: note: suggested alternative: 'I2C_CLK'
     audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, I2S_MCLK);
                                                  ^~~~~~~~
                                                  I2C_CLK
*** [.pio\build\esp32s3\src\audiotask.o] Error 1
================================================================================================== [FAILED] Took 48.99 seconds ==================================================================================================

I am not so practiced in programming, my part is more hardware...so BIG THX in advance for YOUR help ;-)
Best regards
OldTinkerer

schreibfaul1, 2024-01-09:
Thank you for your tip. I have corrected an error during an update. It will work with the current version

OldTinkerer, 2024-01-09:
BIG THX!

Danke Mann, ich jage es morgen durch den Compiler, heute breche ich schon weg, Info folgt ;-)

closed 2024-02-24 by github-actions[bot]