schreibfaul1/ESP32-MiniWebRadio #333: System hangs after "audiotask: priority is 2" (solved) and wrong WiFi credentilas

RStaunlich, 2024-01-10:
Hi,
First of all, huge praise for the entire work! It now runs almost perfectly and stably with the following setup:
- ESP32-DevKitC-VE
- ILI9341
- VS1053

I only had start-up problems at the beginning - in 9 out of 10 cases the system got stuck after "audiotask: priority is 2".
**Solution:** undef LOAD_VS0153_PLUGIN in file \.pio\libdeps\esp32\ESP32-vs1053_ext\src\vs1053_ext.cpp

 The only remaining problem is that the message "WiFi credentilas are not correct" appears at every second start. Does not seem to be a problem with a weak WiFi signal...

RStaunlich, 2024-01-10:
...possible workaround: a simple `while(wifiMulti.run() != WL_CONNECTED) {...} ` with timeout does the job in main.cpp

schreibfaul1, 2024-01-10:
Hello [RStaunlich](https://github.com/RStaunlich),
I have now installed a delay. Depending on the signal strength, the connection may not be established otherwise.

closed 2024-02-25 by github-actions[bot]