schreibfaul1/ESP32-MiniWebRadio #334: No Issue, just need help

chess-levin, 2024-01-14:
Hi, 
I'm trying to migrate your simple example [ESP32-vs1053_ext project](https://github.com/schreibfaul1/ESP32-vs1053_ext) from a ESP32 WROOM DecvKitC4 to a ESP32-S3 DevKitC1 N16R8. I want to use a board with PSRAM, hoping for a bigger buffer and less `DEBUG:  slow stream, dropouts are possible` messages. Unfortunately you closed the old project and so I hope I can place my questions here.

I got the old project running on an ESP32 DecvKitC4, but got stuck when trying to port it to a ESP32-S3. There is this line in your example code `VS1053 mp3(VS1053_XCS, VS1053_XDCS, VS1053_DREQ, VSPI, VS1053_MOSI, VS1053_MISO, VS1053_SCK);` that doesn't compile. It seems that the `const VSPI` isn't defined when I'm using the dev environment for the ESP32-S3. How should I initialize the VS1053 object? I' also have troubles to identify VSPI and HSPI on the [ESP32-S3](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html). There are labels like FSPIHD, SUBSPIHD, SPICLK_N etc. Which could I take for VS1053B, [ILI9488 OLED Display and SD_Card](https://www.amazon.de/dp/B08GBYQQV5)?

I want to build a similar project as your MiniWebRadio, but trying to build and understand it step by step:
1. setting up output of sound
2. integrating display 
3. integration touch function
4. preferences
5. ota

I've some more questions regarding the VS1053B module I' using:

On my VS1053 breakout board is a chip that is label "VS1003B" and not "VS1053B". Afaik is the VS1003 the predesseor of the VS1053B. But your `vs1053_ext.h` library works well. I'm getting sound connecting to e.g. `mp3.connecttohost("http://icecast.radiobremen.de/rb/bremenzwei/live/mp3/128/stream.mp3");`

I'm trying to check the chip's version by calling `mp3.printChipID()` after `mp3.begin()`, but I can't understand the return values. Could you help?

```
  Serial.printf("VS1053 ChipId %lu;%d;%u\n", mp3.printChipID(), mp3.printChipID(),  mp3.printChipID());
  VS1053 ChipId 2422247520;-1872719776;2422247520
```

However I ordered a second breakout board with a chip labeled VS1053B, but unfortunately this board isn't working at all. I connected it the same way as the "VS1003B" board and no sound is played. Any ideas for testing the board?

Thank you for all your work published on github. I appreciate any help.

schreibfaul1, 2024-01-14:
With the ESP32-S3 you can use HSPI as a second SPI bus
 ````VS1053 vs1053(VS1053_CS, VS1053_DCS, VS1053_DREQ, HSPI, VS1053_MOSI, VS1053_MISO, VS1053_SCK);````
The "slow stream" problem with some stations cannot be solved with a memory extension. The real bottleneck is the Arduino WiFi.
https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/Better_WiFi_throughput shows how to fix this.

schreibfaul1, 2024-01-14:
Addendum: VSPI cannot be used with the ESP32-S3 (is used internally) but there is FSPI instead

chess-levin, 2024-01-15:
Thank you for your quick feedback. I've already seen your code for [Better_WiFi_throughput](https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/Better_WiFi_throughput). But I don't understand how to execute it.
I'm using Arduino IDE and mainly PlatformIO, but having no experience with the [Espressif IoT Development Framework](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/). Do I have to install idf parallel to PlatformIO in order to get this Configuration Tool? Is there a idf-plugin for PlatformIO I can use? Or do I just have to checkout your project code and build it with PlatformIO?
I'd really appreciate some more details on how to do this tcp-reconfiguration. Thanks again.

schreibfaul1, 2024-01-15:
You cannot use a precompiled Arduino version in this case.
If you use PlatformIO, you can include Arduino as a component. Here is an example: https://github.com/schreibfaul1/ESP32_Arduino_ESPIDF

chess-levin, 2024-01-15:
Thanks again. 

I've cloned your ESP32_Arduino_ESPIDF project.
Opened the menuconfig and made the new TCP configuration (BTW: As windows user  you have to use the keys 'j','k' to navigate between the menuitems ;-) Finally saved the changes to sdkconfig.esp32dev. Successfully built your project code.

Everything looked fine until here.

Then I  copied my code into main.cpp

```
#include <sdkconfig.h>
#include "Arduino.h"
#include <SPI.h>
#include <WiFi.h>
#include "vs1053_ext.h"

// https://github.com/schreibfaul1/ESP32-vs1053_ext

// Digital I/O used
#define VS1053_XRST   1
#define VS1053_XDCS   4
#define VS1053_XCS    5
#define VS1053_DREQ   36

#define VS1053_MOSI   23
#define VS1053_MISO   19
#define VS1053_SCK    18

String ssid     = "***";
String password = "***";
String hostname = "WEBRADIO_ESP32_WROOM";

int volume=15;

VS1053 mp3(VS1053_XCS, VS1053_XDCS, VS1053_DREQ, VSPI, VS1053_MOSI, VS1053_MISO, VS1053_SCK);

void printEspChipInfo() {
  Serial.println("================ ESP ChipInfo =====================");
  Serial.printf("Chip model %s (rev %d), #cores %d, CPU Freq %d MHz", ESP.getChipModel(), ESP.getChipRevision(), ESP.getChipCores(), ESP.getCpuFreqMHz());
  Serial.printf(", FlashChipSize %d KB, FlashChipSpeed %d\n ", ESP.getFlashChipSize()/1024, ESP.getFlashChipSpeed());
  Serial.println("================ ESP ChipInfo =====================");

  Serial.println("================ SPI GPIO =====================");
  Serial.print("SPI MOSI: ");  Serial.println(MOSI);
  Serial.print("SPI MISO: ");  Serial.println(MISO);
  Serial.print("SPI SCK : ");  Serial.println(SCK);
  Serial.print("SPI SS  : ");  Serial.println(SS);
  Serial.println("================ SPI GPIO =====================");

}

void setup() {
  
  Serial.begin(115200);

  printEspChipInfo();

  SPI.begin();
  //SPI.begin(VS1053_SCK, VS1053_MISO, VS1053_MOSI);

  Serial.print("Connecting to wifi ");
  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  
  WiFi.setHostname(hostname.c_str()); //define hostname
  
  WiFi.begin(ssid.c_str(), password.c_str());
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    delay(1000);
  }
  Serial.print(" success: ");  Serial.println(WiFi.localIP());
  Serial.print("RRSI: ");  Serial.println(WiFi.RSSI());

  Serial.println("mp3.begin");

  mp3.begin();

  Serial.print("VS1053 ChipId "); Serial.println(mp3.printChipID());

  
  //  mp3.loadUserCode(); // FLAC plugin
  mp3.setVolume(volume);
  mp3.connecttohost("http://icecast.radiobremen.de/rb/bremenzwei/live/mp3/128/stream.mp3");
}

// The loop function is called in an endless loop
void loop()
{
    mp3.loop();
}
```

and got a lot of linker errors regarding undefined functions in WiFiClientSecure

```
Dependency Graph
|-- ESP32-vs1053_ext @ 1.0.10+sha.d8a7551
Building in debug mode
Compiling .pio\build\esp32dev\src\main.o
Linking .pio\build\esp32dev\firmware.elf
c:/users/xxx/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\esp-idf\framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa\libframework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa.a(WiFiClientSecure.o):(.literal._ZN16WiFiClientSecure4stopEv+0x4): undefined reference to `stop_ssl_socket(sslclient_context*, char const*, char const*, char const*)'
c:/users/xxx/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\esp-idf\framework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa\libframework-arduinoespressif32-src-52b8cdec3e60619426e51a1d18d1c3fa.a(WiFiClientSecure.o):(.literal._ZN16WiFiClientSecure5writeEPKhj+0x0): undefined reference to `send_ssl_data(sslclient_context*, unsigned char const*, unsigned int)'
```

What did I miss out?

chess-levin, 2024-01-20:
Hello,

now I've cloned your project https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/Better_WiFi_throughput and changed platformio.ini to match my [ESP32 devkitc v4](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-devkitc.html). 

* Compiling is ok
* Linker gives some warnings

But your partition scheme does not match my ESP32 devkitc v4 with 4MB flash memory. I tried to change the values of your `partitions.csv` with some kind of default, but that doesn't work either.

```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 1M
```

So, the final mem check stops the code from being uploaded :-(

```
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Archiving .pio\build\esp32dev\lib9d7\libESP32-audioI2S.a
Linking .pio\build\esp32dev\firmware.elf
Retrieving maximum program size .pio\build\esp32dev\firmware.elf
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
Error: The program size (1225713 bytes) is greater than maximum allowed (1048576 bytes)
RAM:   [=         ]  10.7% (used 35128 bytes from 327680 bytes)
*** [checkprogsize] Explicit exit, status 1
Flash: [==========]  116.9% (used 1225713 bytes from 1048576 bytes)
================================================================= [FAILED] Took 111.22 seconds =
``` 

* Do you have a default `partitions.csv` for my ESP32 DevkitC v4 4MB?
* Do I have to change anything else than the `partitions.csv`?

Thanks again.

schreibfaul1, 2024-01-20:
1M for the factory app is too small. Have a look here to see what PlatfornIO has for predefined boards. https://github.com/platformio/platform-espressif32 
Some compiler warnings are normal, depending on the Arduino version

schreibfaul1, 2024-01-20:
Your board will not have PSRAM, please remove -DBOARD_HAS_PSRAM

chess-levin, 2024-01-20:
The param `-DBOARD_HAS_PSRAM` was prefixed with a comment `;`, but nevertheless, I removed it.  The ESP32 still crashes.

```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#v6.2.0  ; ESP-IDF v4.4.4
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.8

framework = arduino, espidf

board = esp32dev ;chipmodel ESP32, 4M FLASH, USBtoTTL

;board_build.f_cpu = 240000000L
;board_build.flash_size=4MB
;board_build.flash_freq=80M
;board_build.spiram_mode=2
board_build.partitions = partitions.csv

monitor_speed = 115200
monitor_filters = esp32_exception_decoder

upload_speed = 460800  ; 921600, 512000, 460800, 256000, 115200

build_flags =
        -Wall
        -Wextra
        -DCORE_DEBUG_LEVEL=3
        -DCONFIG_ARDUHAL_LOG_COLORS
;        -DAUDIO_LOG
        -DARDUINO_RUNNING_CORE=1       ; Arduino Runs On Core (setup, loop)
        -DARDUINO_EVENT_RUNNING_CORE=1 ; Events Run On Core

lib_deps =
    https://github.com/schreibfaul1/ESP32-audioI2S.git
```

I think it's a wrong setting in the `sdkconfig.esp32dev`

Any hint how to change it?

chess-levin, 2024-01-20:
I tried `[ESP32-MiniWebRadio](https://github.com/schreibfaul1/ESP32-MiniWebRadio/tree/master)
/sdkconfig.esp32] instead and now it's working!!!!

chess-levin, 2024-01-21:
Hello Wolle,
is there a way to avoid the recompilation of the complete arduino files each time I'm building my project? That takes a lot of time. cmake should know that the sources aren't touched. Any hint on this topic?
Thanks again

closed 2024-03-07 by github-actions[bot]