schreibfaul1/ESP32-MiniWebRadio #335: Ni issue,just a question about KCX_BT_Emitter using

clear-sky, 2024-01-15:
Hi dear wolle!
just want to know what is the main reason of using KCX_BT_Emitter module instead of using a nice audio amplifire?
each cell phone has very very better hardware for using it as a web radio with bluetooth sender
i think new design in its best condition is a loser in competition with a mobile cell phone (edzelf tried this design and did not take a good feedback)
https://github.com/Edzelf/ESP32Radio-V2

schreibfaul1, 2024-01-15:
Hello clear-sky,
a BT functionality is desirable. Audio BT is not possible with the ESP32-S3, with the ESP32 only to a limited extent (either WiFi or BT with high resource consumption) and future chips from Espressif will probably not have audio BT either. The KCX_BT emitter is not that bad, but difficult to wire for beginners with little soldering experience (1.25mm contact spacing). I am in the process of writing the routines required for integration, but this will take some time. In addition, the BT_Emitter should be purely optional, so it is not absolutely necessary.

Here are my thoughts on audio amplifiers:
If you buy an amplifier and simply plug it in, it will often be a disappointment. Rattling and cracking noises are unavoidable.
1) Interference from the power supply: Audio and digital parts should be supplied separately, not both via the USB 5V. If there is no other way put RC 27Ohm/100µF everywhere. Always use AGND for audio lines, not GND of the power supply. 
2) HF interference is caused by the WiFi or (and) BT antenna. Therefore, place the audio amplifier as far away as possible from the antennas. However, every piece of wire also acts as a receiving antenna. Shielding measures with sheet metal are not always applicable. Good results can be achieved with antistatic packaging. These are slightly conductive films in which the industry packs electronic components. The film is usually transparent grey.
If  so there are no problems with audio amplifiers and great sound.

schreibfaul1, 2024-01-15:
Then you've done everything right :-)) But there are also some people reading here who don't know the basics of analogue electronics.

clear-sky, 2024-01-15:
and you are professional in radio!
i have seen your great works making an analog fm radio with tubes in youtube !!!!!!
Very few people understand the level of your work!!! 
I envy your level of knowledge
_thanks with respect_

closed 2024-01-15 by clear-sky

clear-sky, 2024-01-17:
hi all
this is a nice video for those users that suffer from white noise 
https://www.youtube.com/watch?v=n_b_0BWP0mI