schreibfaul1/ESP32-MiniWebRadio #336: pin belegung vjmci 5202

maba64, 2024-01-15:
[image]

Hallo
ich habe eine Frage betr dem 5202 dac modul
welche pins benötige ich da ?
verwende den esp32 s3  n16r8
besten Dank für deine Antwort
Markus

JozefKc, 2024-01-18:
Hi Markus!
wegen diesen Sachen habe ich in Issue 71 /  https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/338  / Scheme zugelegt. Danach musst man nur die AnalogausgängeL, R und G /ground/ yum Vestärker anschliessen.
Viel Glueck!

maba64, 2024-01-18:
besten Dank

maba64, 2024-02-02:
hallo
leider bekomme ich keinen Sound 😥
hast du betr Din und dout bei den pin def etwas angepasst ?
Gruss Markus

JozefKc, 2024-02-02:
Hallo Markus,
ich habe das wieder gecheck:
PCM5102 <---> Processor
-------------------------------------------------
Vin            auf    +3,3V
GND         auf    GND /heisst 0V/
SCK         auf     GND anschliessen
Din           auf     GPIO9
BCK         auf     GPIO3
L_RCK     auf     GPIO1

Check noch die richtige Definition der PCM5102 durch die Bruecken H1L bis
H4L - sehe Foto 20240202_192318 in der Beilage.
Meine Karte habe ich beim Aliexpress gekauft und funktioniert
problemloss.....
Wird Du irgendwelche Probleme damit haben - schreib mir.
Viele Erfolge!
Jozef

maba64, 2024-02-02:
hallo
Besten Dank
kannst Du bitte noch das Foto senden?
Dankeschön
Markus

maba64, 2024-02-03:
besten Dank
es lag an den fehlenden Lötbrücken
nun funktionierts
Gruss Markus

JozefKc, 2024-02-03:
Hallo Markus,
ich bin jetzt ganz froh dass alles funktioniert. Viel Spass.
Jozef

closed 2024-02-10 by maba64