schreibfaul1/ESP32-MiniWebRadio #338: Scheme for ESP32-S3-WROOM1 N16R8 and IR code problem

JozefKc, 2024-01-18:
Hello Wolle!
Thank you for your great job. After many experiments I finaly decided to use **ESP32-S3-WROOM1 N16R8** because of good price, sufficient power, pins and simplificity to connect the Wifi network.
I use **ILI9431** with touch from Aliexpress. This touch area works with **HX8347D** driver and TP rotation 3. I had to change a bit the tft.cpp programm - works flawlessly. For all of them, who want to save their time I want to add a scheme I used for my radio. VS1053 is the old version - I do not use it anymore because of trublem with freing of the system at the start, unusable VU meter and problem to interprete som formats like ogg.... I do prefere PCM5102 now.
IR works also fine. I wanted to buy Xindex   IR, but it is not accessible. I am using a similar one - there is a problem. I tried to change these code in main.cpp - row 470-493 with new codes and nothing changed. Could you suggest me any solution how to change it?
Best regards
Jozef

[image]

[image]

schreibfaul1, 2024-01-18:
There is a configuration menu on the website under "IR Settings"
[image]. 
This allows you to assign functions to the buttons [image]

JozefKc, 2024-01-22:
Thank you for your support. I did not merge it.....

closed 2024-03-08 by github-actions[bot]