schreibfaul1/ESP32-MiniWebRadio #339: Streamtitle is not displayed

TubaFonse, 2024-01-19:
Dear Wolle, 

the Streamtitle is not displayed in the current software release. 
The Streamtitle is also missing in serial monitor. Only the station name is displayed permanently. 

Please help - many thanks, 
Alfons.

schreibfaul1, 2024-01-20:
I think I had terminated the Arduino loop() prematurely. The stream titles are displayed again. There will be a few more updates in the 3.0 branch until V3 becomes master. Many new features will soon only affect the BT emitter until it is added to the website and the display.

fonzie15, 2024-01-23:
Hi Wolle!
Looking forward to the code update. Just FYI: I've observed the same issue when playing Internet Radio, but it happened just in some cases. 

Example URLs which work:
https://orf-live.ors-shoutcast.at/fm4-q2a
http://live.antenne.at/arr

Example URLs which don't work (but basically they are working - tested with VLC on PC):
https://stream1.liferadio.at 
http://s1-webradio.webradio.de/rockantenne-oesterreich

Hope that helps..

schreibfaul1, 2024-01-23:
Hi fonzie15,
thanks for your contribution. The example links have helped. I was able to localise the error and eliminate it in the audioI2S library. For an update, you can reload the library in PlatformIO with "Full Clean".

Best regards
Wolle

TubaFonse, 2024-01-24:
Many thanks @schreibfaul1 and @fonzie15 for solving the "streamtitle" issue. 
I'm just tested two URLs in my station list with actual V3, which make further problems (both are ok with last V2 master release): 

http://stream.laut.fm/zusamstream  
https://hls-01-europaplus-top40.emgsound.ru/21/playlist.m3u8 
problem in V3 only : [701253][E][Audio.cpp:3508] parseHttpResponseHeader(): timeout 

This station often sends long streamtitle, but cutted off in display (V3 and also in V2: txt overflow)
http://edge05.streams.arabella.at/arabella-austropop
19:45:53 AUDIO_info:  SampleRate: 44100
19:45:53 AUDIO_info:  BitsPerSample: 16
19:45:53 AUDIO_info:  BitRate: 59087
19:45:53 AUDIO_info:  AAC HeaderFormat: ADTS
19:45:53 AUDIO_info:  AAC Codec: MPEG-4 LowComplexity
19:45:53 StreamTitle: WERNER ZAUNER - EINE GRILLE AUF DER BRILLE
19:45:53 AUDIO_info:  VBR recognized, audioFileDuration is estimated
19:45:53 Audiocodec:  AAC
19:45:54 txt overflow, winHeight=128, strlen=42, written=36, str=WERNER ZAUNER - EINE GRILLE AUF DER BRILLE
19:46:30 StreamTitle: MICHAEL SEIDA - WANN MEI MORGEN NIMMA KUMMT
19:46:31 txt overflow, winHeight=128, strlen=43, written=32, str=MICHAEL SEIDA - WANN MEI MORGEN NIMMA KUMMT

do you have a solution? 

Many thanks, 
Alfons

closed 2024-03-10 by github-actions[bot]