schreibfaul1/ESP32-MiniWebRadio #340: KCX-BT-EMITTER V1.4  MiniWebRadio V3.0 Intial Test

ve2cfb, 2024-01-22:
Hello Wolle,

Did an initial test and here are my Hardware info. Had a couple BT that where DOA.
1- Since V2 of MiniWebRadio i was using a KCX-BT V1.2(PCB). The Led pin was used for a front button with a led.
2- With V3 of MiniWebRadio  KCX-BT is V1.4 and V1.7(PCB). The Led pin has become a LINK  control pin.
3- I intend to use the LED on the KCX-Bt . A small change will do the trick.
3- Had to change my main PCB board to reflect those changes. 

Once i got every working, i ran a utility to get the MAC/Name and added 2 BT Speaker to the Web Page.
One Earphone (Avantree AS50) worked perfectly, it paired within less 10 sec.
My other Speaker is a Sony KD-75X85K. This one is problematic. Sometime it paired and some other time, nothing.
For now, i have to get the MAC / NAME from a utility.  I guess Wolle with make an update to the web Page.

So Wolle, it look promising. Many Thanks.
Pierre

ve2cfb, 2024-01-25:
Hi Wolle,
The biggest problem i am having, is  using some  utilities to read the KCX-BT.
I want to make sure that i have good module before using MiniWebRadio.
So far, i have 4 modules and they all have the same result (not consistant).
I must be doing something really wrong for sure.
I am using a breadboard and i was afraid that the wires where too long. 

I am using Arduino V2 for testing with a UNO NANO.
Today, i will test with a ESP32.

**Simple utility:**
[KCX_BT_Simple_Utility_V1.0.zip](https://github.com/schreibfaul1/ESP32-MiniWebRadio/files/14051627/KCX_BT_Simple_Utility_V1.0.zip)

**Advance utility:**
[KCX_BT_EMITTER_Advance_TEST_V1_02.zip](https://github.com/schreibfaul1/ESP32-MiniWebRadio/files/14051628/KCX_BT_EMITTER_Advance_TEST_V1_02.zip)

The advance utility, can do all AT command but the result is inconsistant. Doing a SCAN command will sometime show my headphone and some other time will not.

The Simple utility, start scanning every sedond or so by itself. 
And i cannot do a AT+ command.

Maybe i need a reset type command before starting.
Maybe you do have some utility that make sense.

Thanks for your time.

schreibfaul1, 2024-01-25:
Hello Pierre,
the version I have published today is the first one that works with the BT emitter at all. If no names or addresses of the BT devices are stored, the BT emitter always connects to the first device scanned. If names or addresses are saved, it will only connect to these.  If the BT emitter is not connected, it is in SCAN mode. All BT devices that have been found are displayed on the website in the lower area.
 [image] 
You can save the value with the "+" buttons. The "BT Mode" button switches between TX<->RX. Volume, pause does not yet work. This is what the log looks like when a device is connected.
[image]

ve2cfb, 2024-01-25:
Hi Wolle,
Thank you for the update.
I will try it this afternoon and give you some feedback.
Pierre

schreibfaul1, 2024-01-25:
The website with all its functions is now ready.
[image]

ve2cfb, 2024-01-25:
wow... quite a surprise...
This is great.
 
2 BT device detected, did connect to my AVANTREE AS50 without any problem.
The Sony one did not but its a bit fussy. i would not spend too much time on this one.
Will do more test later.
Pierre

ve2cfb, 2024-01-31:
Hello Wolle,
I got 3 Speakers working perfectly with your program and i tried the receive part of KCX-BT module and with my Android phone, was able to connect and play some music.  Good work.

Here is my new Breakout board. Presently i am not using the Blue Led.

[image]

I have a problem when shutting power off the BT module.
Still working ont it.

Many thanks for the good work.

ve2cfb, 2024-02-02:
Hello Wolle,
Here is a small Video of my setup.
The KCX-BT module has 2 power Switch.
The top SW is the radio power SW.
 and the Bottom Switch, is for the BT and it has a bi-polar led(Red/Blue).
On the video, the BlueTooth is in Emitter Mode.
https://github.com/schreibfaul1/ESP32-MiniWebRadio/assets/13633451/6cf0195a-faef-4f44-9335-4f2420d7aa27

Everything works ok but with a BT Switch, it is different in some situation.

1- After power on (Radio), with BT switch off,  its impossible to activate 
    the BT by pressing the BT SW to ON. No RED light and no pairing.
2- Next Scenario, before Starting the Radio (Green Button) , initialy the BT SW has to be in ON mode.
     In this mode everything works but if i open the BT SW, the light still blinking but i loose the pairing
     even pressing the BT SW back to ON, it does not work. Must be a control line .
3-  Redo a Power reset and eveything is back to normal. 

Side Note:

By the way, i found a good function from your radio...
You probably know this function.

Here is the sequence... (i will try to make it simple).
1- start radio with a BT speaker
2- In the MiniWebRadio, Go to your Bluetooth menu.
    In that menu, BT goes in receive mode 
    but the light is still RED and should be BLUE.
    (side note... In the web page, going from emitter to receive
    the led does flip from RED to BLUE).
3- In BT Receive mode, start your Android Telephone,
    In Android bluetooth menu,  select MiniWebRadio.
    In youtube.com, play some music.
4- Now you have your Android sending music to ESP32,
    and ESP32 sending android music to your Bluetooth speaker.

Pretty good.

ve2cfb, 2024-02-12:
Hello Wolle,
For now, i will disable the BT switch and only show the Bipolar LEd.
If i need to reset the BT, i will power off the Radio and that will do the trick.
Thanks again...  Pierre

closed 2024-02-17 by ve2cfb