schreibfaul1/ESP32-MiniWebRadio #341: In V3 ampamp in DNLA URL

Linkenelis, 2024-01-29:
DNLA could not play the file (unauthorized was the message)

In Audio.cpp I put these lines to correct:
bool Audio::connecttohost(const char* host, const char* user, const char* pwd) {
    // user and pwd for authentification only, can be empty

    xSemaphoreTakeRecursive(mutex_audio, portMAX_DELAY);

    if(host == NULL) {
        AUDIO_INFO("Hostaddress is empty");
        stopSong();
        xSemaphoreGiveRecursive(mutex_audio);
        return false;
    }

//Linke Nelis: 4 lines to get rid of "ampamp" in url
String amp="";
    if(lastIndexOf(host, 'ampamp') > 1) {
        amp = String(host);  //make it possible to replace in String amp
        amp.replace("ampamp", "") ; //replace the ampamp by empty
        host = amp.c_str();}    //put new String in host

schreibfaul1, 2024-01-29:
Hi @Linkenelis,
that's strange, where does this "ampamp" come from?
If so, I would suppress it in the DLNA lib, not in audioI2S.
Is there any way I can reproduce this?

schreibfaul1, 2024-01-30:
I thought again, it may be the HTML character ampersand. Has it been removed with the new [ESP32-DLNA-Client](https://github.com/schreibfaul1/ESP32-DLNA-Client)?

Linkenelis, 2024-01-30:
// Created on: 30.11.2023
// Updated on: 24.01.2024

I think this is the newest available to me?
But I wil delete it from libdeps, so it downloads the latest.
The one in ESP32-DLNA-Client -> src is the same, and is the one used. The one in ESP32-DLNA-Client (root) is older, so I deleted it.
No change...

yes, there are always '&' in front, so &ampamp (& = ampersant) too much of a coincidence. But very strange indeed. Not a clue where this comes from.
when using Debug level 4, I can see it ends with '&amp' and next line is 'amp'. Maybe that gives you a clue?

Tried to fix it inDLNA_Client readContent() first , but found it complicated. Then quick and dirty in Audio.cpp, just to see if it works.
I'll give it another go.

Linkenelis, 2024-01-30:
in DLNAClient.cpp I found this:

browseResult

        if(startsWith(content, "/container")) {
        //    log_i("%s", m_chbuf);
            item1 = false;
            uint16_t cNr = m_srvContent.size;
            makeContentPushBack();
            replacestr(m_chbuf, "&quot", "\"");
            replacestr(m_chbuf, "&ampamp", "&");   // ampersand
            replacestr(m_chbuf, "&ampapos", "'");  // apostrophe
            replacestr(m_chbuf, "&ampquot", "\""); // quotation

So the fix is there, for starts with:
/container
/item

schreibfaul1, 2024-01-30:
I have now been able to reproduce the error. The original was the ESP32-SOAP library, which I replaced with my own due to its size. The three  ' & "  characters have special encodings in XML. But now it works for me. [image] 
Please reload the external libraries with "Full Clean".

Linkenelis, 2024-01-30:
compiling

Linkenelis, 2024-01-30:
Made all the difference!

Thanks, I learned someting

closed 2024-02-01 by Linkenelis