schreibfaul1/ESP32-MiniWebRadio #342: Not an issue - just an option for using different screen

z34pec, 2024-01-30:
Hello everyone.
I have TZT 1.8 inch TFT LCD Module LCD Screen Module SPI serial 51 drivers 4 IO driver TFT Resolution 128*160 . I wonder how can I use this screen with this version of esp32 web radio ???
Is there any possibility for new workout using 128*160 and 1.8 inch TFT screen ?
It is ST7735S driver on it.
I hope I will receive fas answer :)
Bye.

schreibfaul1, 2024-01-30:
Such a small display is not supported, only this one:
[image]

z34pec, 2024-01-31:
OK, so I have to buy a new display :( . Although I suppose driver for this display is the same as ST7796 because searching for info about ST7735S i managed to find a site where is a driver for all displays from this series (ST77xx) .

schreibfaul1, 2024-01-31:
The resolution is too low for the graphics used

closed 2024-03-16 by github-actions[bot]