schreibfaul1/ESP32-MiniWebRadio #343: Issue with using Arduino ESP32 board

francoispap, 2024-01-31:
Hello,
Thanks for the code
I am trying to run your simple code where the Issues page is now closed for new requests
I am using the Arduino nano esp32 board and I am getting  just one error when compiling (see below)
VS1053 mp3(VS1053_CS, VS1053_DCS, VS1053_DREQ, VSPI, VS1053_MOSI, VS1053_MISO, VS1053_SCK);
VSPI is not recognized
I have changed to the Wroom board and it works fine when compiling
Don't know that cause this one the Arduino board

Please can you help on that ?
Thanks
Pierre

schreibfaul1, 2024-02-01:
It is possible that VSPI is used internally, alternatives are HSPI or FSPI

francoispap, 2024-02-04:
Thanks, I tried what you suggested and thats OK now
There is an On board flash memory connected to SPI

closed 2024-03-21 by github-actions[bot]