schreibfaul1/ESP32-MiniWebRadio #345: A problem with a mirror touch screen display

tzvetan65, 2024-02-04:
Hi Wolle
An excellent project, everything works perfectly and is very functional. However, there is one problem with which I do not find a solution - the real place on the buttons is mirrored on their graphics. For example, the chart of the button is on the lower left corner and the display accepts this command In the lower right corner. The display is 3.2 TFT SPI 240x320 V1.0, the display driver is Ili9431, the touchscreen driver is HX8347D, TFT Rotation 3, TP rotation 1.
Please for help, thank you in advance.

tzvetan65, 2024-02-04:
I missed writing, the chip is ESP32 Wroom 32D, DAC is VS1053B

steve6375, 2024-02-04:
Did you try other Rotations ? e.g. TFT Rotation 1, TP rotation 3

tzvetan65, 2024-02-04:
Yes, I tried this option as well, but the touch buttons are still mirrored on the graph.

closed 2024-02-06 by tzvetan65

steve6375, 2024-02-06:
There should be an option added for this - e.g. _rotation == 5  or something ???

tzvetan65, 2024-02-06:
Hi, I did it in rotation = 2, changing "x = y" with "x = HX8347D_HEIGHT - y" and so I compiled it. There may be reason to add a rotation 5 =, but this must be decided by the author of the project.

schreibfaul1, 2024-02-07:
This is a rare display. With the usual displays, one rotation was always enough. To cover all cases I will add a horizontal and vertical mirror.  @tzvetan65 do you have a link to your display so I can see how it looks like.