schreibfaul1/ESP32-MiniWebRadio #347: SD Card Mount Fail

jackmiller1986, 2024-02-06:
Hi Guys, 

Iam since few days on the project .

Iam failing about the sd card Mount fail.

I have an ILI9844 where can i connect the sd port on the display itself, what are the pins on the esp32. 

Because the display has SD_CS, SD_MOSI, SD_MISO, and SD_SCK. Where should go this pins from the display sd to the esp32. I cant get sd to work.

schreibfaul1, 2024-02-06:
Die Pins haben eine andere Bezeichnung
[image]

jackmiller1986, 2024-02-06:
ja wohin kommt denn der cs.

Habe nun am display folgendes an den sd port,

MOSI= GPIO15
MISO=GPIO2
SCK=GPIO14

Und woran muss CS???? Wie gesagt habe ein ILI9488 mit SD Schacht am display das mag ich nutzen.

jackmiller1986, 2024-02-06:
Okay auf den normalen esp32 scheint es zu laufen nun.

closed 2024-03-22 by github-actions[bot]