schreibfaul1/ESP32-MiniWebRadio #348: ESP32-S3 Hilfe Diagramm

jackmiller1986, 2024-02-06:
Hi habe nun geschafft es auf einen normalen ESP32 zum laufen zu kriegen.

Benötige aber nun hilfe da ich im netz kein brauchbares Diagramm finde.

Ich habe diesen esp32 mit PSRAM noch hier: https://www.amazon.de/dp/B0CMNK1Q2H?psc=1&ref=ppx_yo2ov_dt_b_product_details

Ein ESP32S Mit 16mb Flash und 8 MB PS ram.

Bräuchte ein Diagram um das ILI9844 dran anzuklemmen und was ich beim Flashen dessen beachten muss,

Arne65, 2024-02-06:
Ja das kannst du so benutzen 

1. unter plattformio.ini in VS Code must du dein Board auswählen 

[env:esp32s3]
        ;board_build.partitions = miniwebradio4MB.csv    ;  4MB Flash
        ;board_build.partitions = miniwebradio8MB.csv    ;  8MB Flash
        board_build.partitions = miniwebradio16MB.csv   ; 16MB Flash

        ;board = ESP32-S3-DevKitC-1-N8R2        ;  8MB Flash, 2MB quad PSRAM +1)
        ;board = esp32-12k (N4R4)               ;  4MB Flash, 4MB quad PSRAM +1)
        ;board = ESP32-S3-DevKitC-1-N8R8        ;  8MB Flash, 8MB octo PSRAM  *2)
        board = ESP32-S3-DevKitC-1-N16R8       ; 16MB Flash, 8MB octo PSRAM  *2)

2.  unter PROJEKT TASKS ---> ESP32s3 ----> General -----> Run Menuconfig
 

   ESP32S3-Specific ---> SPI RAM config ---> MODE (QUAD/OCT)... ---> (x) Octal Mode PSRAM

einstellen

jackmiller1986, 2024-02-06:
unter PROJEKT TASKS ---> ESP32s3 ----> General -----> Run Menuconfig
ESP32S3-Specific ---> SPI RAM config ---> MODE (QUAD/OCT)... ---> (x) Octal Mode PSRAM

einstellen. Wo finde ich das genau??

jackmiller1986, 2024-02-07:
Also Der esp32 von Amazon mag das wohl nicht so , die hintergrundbeleuchtung von ili9488 geht an , also schwarzes display. Aber display beleuchtung an. Aber Bootet es nicht anscheinend. Kein plan ob es am billig clone liegt. Habe jetzt mal von Berrybase nen Originalen bestellt mit 8mb flash und 8 mb PSram . 16 mb hatten die da nicht nur 32 mb. Denke mal das wird mit der kombi 8 mb flash und 8mb psram auch lüppen. Da der normale esp32 ohne psram echt mit dem image zu kämpfen hat.

Arne65, 2024-02-07:
Wichtiger als das Display (das Display zeigt nur etwas an wenn der ESP wenn das Programm läuft) 
 Was steht im Terminal von VS Code an Meldungen ?

Wenn du den PSRAM unter RunMenuconfig nicht auf OCTAL getsellt hast wirst du keinen Erfolg haben .

[image]
[image]

Arne65, 2024-02-07:
Prüfe im seriellen Monitor was eigentlich passiert

jackmiller1986, 2024-02-07:
der fehler ist gefunden der clone esp32 gibt keine 5v raus auf 3,3 volt geht es nun

jackmiller1986, 2024-02-07:
so muss nur noch herauskriegen wie ich die logos ändere und in welchen format.

jackmiller1986, 2024-02-07:
So läuft alles nun , mal ne frage kann man die std ansage abstellen??

closed 2024-03-23 by github-actions[bot]