schreibfaul1/ESP32-MiniWebRadio #350: DLNA Issue in V3

TubaFonse, 2024-02-18:
Hallo Wolle! 
Ich bin am Testen von V3.0 und warte schon sehnsüchtig, bis endlich mein KCX_BT_Emitter mit der Post kommt ... 

In der aktuellen V3.0 Version kann ich derzeit über DLNA keine Musik mehr abspielen! 
Nach Auswahl des Titels (über Select level 4) erhalte ich folgende Meldung (und eine endlose Boot-Schleife): 
[image]

17:16:24 DLNA_server: title Wunder der Musik, childCount 0, itemSize 17272304, duration 0:03:18
17:16:24 DLNA_server: returned 0 from 0
17:17:19 AUDIO_info:  Connect to new host: "http://   .    .   .  :10243/WMPNSSv4/3176339685/1_N18zODhhNmM4Yi00NjMwMQ.mp3"
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core  1 register dump:
PC      : 0x400570bc  PS      : 0x00060c30  A0      : 0x820a06e2  A1      : 0x3fcbbe00
A2      : 0xbcc37272  A3      : 0x00000000  A4      : 0x0000414d  A5      : 0xbcc37272  
A6      : 0x00060e23  A7      : 0x00000000  A8      : 0x8037a348  A9      : 0x3fcbbdc0
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000008  A13     : 0x3d804288  
A14     : 0x3fca6e6c  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001d
EXCVADDR: 0xbcc37272  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  
Backtrace: 0x400570b9:0x3fcbbe00 |<-CORRUPTED
  #0  0x400570b9:0x3fcbbe00 in ?? ??:0
ELF file SHA256: 96f5f4d78d93e641
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420c91d2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   816][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
ESP32 Chip: ESP32-S3
Arduino Version: 2.0.14
ESP-IDF Version: 4.4.6
MiniWebRadio     Version 3.00k Feb 11/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 8384759 bytes
RESET_REASON: Software reset due to exception/panic
16:17:19    
16:17:19        ***************************    
16:17:19        *     MiniWebRadio V3     *    
16:17:19        ***************************    

Hilft Dir das weiter und kannst Du den Fehler bei Dir reproduzieren? 
Danke & Gruß aus Bayern, 
Alfons

schreibfaul1, 2024-02-19:
Hallo Alfons,
das ist nicht einfach zu ergründen. Ich habe in der heutigen Version "Reset Reasons" hinzugefügt. Damit sollte es möglich sein, bei einem solchen Fehler nicht mehr in eine Endlosschleife zu laufen. Reset-Reasons laufen auf den S3 ganz gut, auf dem ESP32 stimmt das was zurückgemeldet wird nicht ganz, aber es wird trotzdem funktionieren.
Kommt der beschriebene Fehler immer in Verbindung mit dem DLNA Client?  Immer, egal was du auswählst?

Die BT Emitter kommen aus China, die Lieferzeit war bei mir 4 Wochen. Ich hatte glücklicherweise gleich zwei bestellt, der erste ist beim Einlöten kaputt gegangen, vermutlich durch eine Berührung zwischen den Pins.

TubaFonse, 2024-02-19:
Hallo Wolle, 
der beschriebene Fehler kommt nur in der V3 immer in Verbindung mit dem DLNA Client, egal welchen Titel ich auswähle oder auf welchen DLNA Client ich zugreife (ich hab 2 davon). 
Mit der V2 läuft das Abspielen auf meinen DLNA Clients ohne Problem. 
Mit der aktuellen Reset-Reasons Version läuft der Fehler nicht mehr in der Endlosschleife, sondern "hängt" sich wie folgt auf (noch bevor ich einen Titel auswählen kann):  

load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   810][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
ESP32 Chip: ESP32-S3
Arduino Version: 2.0.14
ESP-IDF Version: 4.4.6
MiniWebRadio     Version 3.00m Feb 19/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 8384743 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************
00:00:00        *     MiniWebRadio V3     *
00:00:00        ***************************
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ST7796
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 1897.4 MB by 1938.5 MB free
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  2658][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: FRITZ!Box7490LF
[  7317][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  7317][I][WiFiMulti.cpp:119] run(): [WIFI] 5 networks found
[  7318][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: C8:0E:14:0C:93:11 SSID: FRITZ!Box7490LF Channel: 11 (-45)
[  7367][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 202 - AUTH_FAIL
[  7474][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
00:00:06 WiFI_info:   Connecting WiFi...
00:00:06 WiFI_info:   mDNS name: MiniWebRadio
00:00:06 setup: ....  connected to FRITZ!Box7490LF, IP address is 192.168.178.31
00:00:06 ftpServer:   Buffers allocated: 4622 bytes
00:00:06 rtime_info:  Initializing SNTP
00:00:06 rtime_info:  Waiting for system time to be set... (1/10)
00:00:08 rtime_info:  Waiting for system time to be set... (2/10)
00:00:10 rtime_info:  Waiting for system time to be set... (3/10)
21:02:52 rtime_info:  21:02:52
21:02:52 AUDIO_info:  commFMT = 1
21:02:52 audiotask:   is pinned to core 1
21:02:52 audiotask:   priority is 2
21:02:52 setup: ....  Number of saved stations: 44
21:02:52 setup: ....  current station number: 7
21:02:52 setup: ....  current volume: 20
21:02:52 setup: ....  last connected host: http://edge05.streams.arabella.at/arabella-austropop
21:02:52 setup: ....  connection timeout: 500 ms
21:02:52 setup: ....  connection timeout SSL: 2000 ms
21:02:52 action: ...  current volume is 20
21:02:52 action: ...  switch to station 7
21:02:52 AUDIO_info:  Connect to new host: "http://edge05.streams.arabella.at/arabella-austropop"
21:02:52 AUDIO_info:  PSRAM found, inputBufferSize: 638965 bytes
21:02:52 AUDIO_info:  buffers freed, free Heap: 193356 bytes
21:02:52 AUDIO_info:  Connection has been established in 20 ms, free Heap: 193356 bytes
21:02:52 icy-descr:   Arabella Austropop
21:02:52 StationName: Arabella Austropop
21:02:52 icy-url: ..  https://www.arabella.at
21:02:52 AUDIO_info:  AACDecoder has been initialized, free Heap: 193356 bytes , free stack 5084 DWORDs
21:02:52 lastURL: ..  http://edge05.streams.arabella.at/arabella-austropop
21:02:52 AUDIO_info:  stream ready
21:02:52 AUDIO_info:  syncword found at pos 0
[... 91 lines cut ...]
20:04:17 ftpServer:   Buffers allocated: 4622 bytes
20:04:17 rtime_info:  Initializing SNTP
21:04:17 rtime_info:  21:04:17
21:04:17 AUDIO_info:  commFMT = 1
21:04:17 audiotask:   is pinned to core 1
21:04:17 audiotask:   priority is 2
21:04:17 setup: ....  Number of saved stations: 44
21:04:17 setup: ....  current station number: 7
21:04:17 setup: ....  current volume: 20
21:04:17 setup: ....  last connected host: http://edge05.streams.arabella.at/arabella-austropop
21:04:17 setup: ....  connection timeout: 500 ms
21:04:17 setup: ....  connection timeout SSL: 2000 ms
21:04:18 action: ...  current volume is 20
21:04:18 RESET_REASON:Reset reason can not be determined
21:04:18 WiFI_info:   RSSI is -44 dB
21:04:18 ftpServer:   Ftp server waiting for connection on port 21
21:04:20 BT-Emitter:  KCX_BT_Emitter not found 
[ 12989][E][DLNAClient.cpp:155] srvGet(): The server 192.168.178.24:50001 is not responding after request
[ 12989][E][DLNAClient.cpp:744] loop(): error in srvGet

Hier bleibt er dann "hängen".
Könntest Du bitte mal ausprobieren, ob Du in der aktuellen V3 Version von einem DLNA Client in Deinem Home Netz etwas abspielen kannst? 

Danke & Gruß, 
Alfons

schreibfaul1, 2024-02-19:
Super, mit diesem Log kann ich was anfangen und kann den Fehler reproduzieren. Mir ist nur noch nicht ganz klar, was die Ursache ist.

TubaFonse, 2024-02-19:
In V3 sieht meine Auswahlmöglichkeit so aus: 
[image]
Ausgabe: 
[image]
In V2 sieht die Auswahl im Level 1 viel detaillierter aus:
[image]

TubaFonse, 2024-02-19:
Vielleicht passen da die Übergabeparameter nicht mehr zusammen, was den Crash verursacht?

TubaFonse, 2024-02-20:
Hallo Wolle, 
habe es soeben ausprobiert - über das Web-Interface funktioniert die Auswahl bis zum Level 1, mehr nicht. 
Über den Touchscreen kann ich zumindest jetzt einen Titel auswählen, bis das Programm wie folgt abstürzt und neu bootet: 

19:24:38 DLNA_server: [0] 192.168.178.29:2869  WIN10ALF: Hoeckmayr:
19:24:39 DLNA_server: [1] 192.168.178.24:50001  DS213J_AL
19:24:39 DLNA_server: 2 media server found
19:24:56 txt overflow, winHeight=128, strlen=39, written=36, str=DIE WILDEN KAISER - WARUM REGGAE MI AUF
19:25:08 txt overflow, winHeight=128, strlen=39, written=36, str=DIE WILDEN KAISER - WARUM REGGAE MI AUF
19:25:27 StreamTitle: GEORG DANZER & WILFRIED - SCHAU SCHATZI
19:25:30 DLNA_server: title Bilder, childCount 7, itemSize 0, duration ?
19:25:30 DLNA_server: title Musik, childCount 10, itemSize 0, duration ?
19:25:30 DLNA_server: title Videos, childCount 8, itemSize 0, duration ?
19:25:30 DLNA_server: title Wiedergabelisten, childCount 2, itemSize 0, duration ?
19:25:30 DLNA_server: returned 0 from 0
19:25:38 DLNA_server: title Alben, childCount 2, itemSize 0, duration ?
19:25:38 DLNA_server: title Albuminterpreten, childCount 3, itemSize 0, duration ?
19:25:38 DLNA_server: title Alle Interpreten, childCount 3, itemSize 0, duration ?
19:25:38 DLNA_server: title Alle Musikdateien, childCount 130, itemSize 0, duration ?
19:25:38 DLNA_server: title Bewertung, childCount 5, itemSize 0, duration ?
19:25:38 DLNA_server: title Genre, childCount 3, itemSize 0, duration ?
19:25:38 DLNA_server: title Komponisten, childCount 17, itemSize 0, duration ?
19:25:38 DLNA_server: title Mitwirkende Interpreten, childCount 3, itemSize 0, duration ?
19:25:38 DLNA_server: title Musikwiedergabelisten, childCount 2, itemSize 0, duration ?
19:25:38 DLNA_server: returned 0 from 0
19:26:03 DLNA_server: title Musikgespür, childCount 30, itemSize 0, duration ?
19:26:03 DLNA_server: title Piano Piano, childCount 100, itemSize 0, duration ?
19:26:03 DLNA_server: returned 0 from 0
19:26:12 DLNA_server: title Am Böhmengässchen, childCount 0, itemSize 7295776, duration 0:03:02
19:26:12 DLNA_server: title Am Böhmengässchen, childCount 0, itemSize 16991920, duration 0:03:02
19:26:12 DLNA_server: title Am Kreuzersteg, childCount 0, itemSize 6327988, duration 0:02:38
19:26:12 DLNA_server: title Am Kreuzersteg, childCount 0, itemSize 14131773, duration 0:02:38
19:26:12 DLNA_server: title Astronauten Marsch, childCount 0, itemSize 5832088, duration 0:02:25
19:26:12 DLNA_server: title Astronauten Marsch, childCount 0, itemSize 14041587, duration 0:02:25
19:26:13 DLNA_server: title Blasmusik im Herzen, childCount 0, itemSize 16761056, duration 0:03:12
19:26:13 DLNA_server: title Blasmusik im Herzen, childCount 0, itemSize 7683100, duration 0:03:11
19:26:13 DLNA_server: title Das Tor der Alpen, childCount 0, itemSize 8298016, duration 0:03:27
19:26:13 DLNA_server: returned 0 from 0
19:26:19 AUDIO_info:  Connect to new host: "http://192.168.178.29:10243/WMPNSSv4/3176339685/1_N18zODhhNmM4Yi00NjMwMQ.mp3"

assert failed: heap_caps_free IDF\components\heap\heap_caps.c:381 (heap != NULL && "free() target pointer is outside heap areas")

Backtrace: 0x40377b66:0x3fcbbca0 0x4037e575:0x3fcbbcc0 0x40387ff5:0x3fcbbce0 0x40378ad6:0x3fcbbe10 0x4037a345:0x3fcbbe30 0x420a1d76:0x3fcbbe50 0x420aae68:0x3fcbbe70 0x420646df:0x3fcbbe90 0x420627b4:0x3fcbbeb0 0x4202147b:0x3fcbbed0 0x420218f3:0x3fcbbef0 0x42003599:0x3fcbbf60

  #0  0x40377b66:0x3fcbbca0 in panic_abort at C:/Users/Hoeckmayr/.platformio/packages/framework-espidf@3.40406.240122/components/esp_system/panic.c:408
  #1  0x4037e575:0x3fcbbcc0 in esp_system_abort at C:/Users/Hoeckmayr/.platformio/packages/framework-espidf@3.40406.240122/components/esp_system/esp_system.c:137      
  #2  0x40387ff5:0x3fcbbce0 in __assert_func at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf@3.40406.240122\components\newlib/assert.c:85
  #3  0x40378ad6:0x3fcbbe10 in heap_caps_free at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf@3.40406.240122\components\heap/heap_caps.c:381
      (inlined by) heap_caps_free at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf@3.40406.240122\components\heap/heap_caps.c:366
  #4  0x4037a345:0x3fcbbe30 in esp_mbedtls_mem_free at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf@3.40406.240122\components\mbedtls\port/esp_mem.c:46    
  #5  0x420a1d76:0x3fcbbe50 in mbedtls_free at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf@3.40406.240122\components\mbedtls\mbedtls\library/platform.c:66
  #6  0x420aae68:0x3fcbbe70 in mbedtls_ssl_free at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf@3.40406.240122\components\mbedtls\mbedtls\library/ssl_tls.c:6761
  #7  0x420646df:0x3fcbbe90 in stop_ssl_socket(sslclient_context*, char const*, char const*, char const*) at C:/Users/Hoeckmayr/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/ssl_client.cpp:336
  #8  0x420627b4:0x3fcbbeb0 in WiFiClientSecure::stop() at C:/Users/Hoeckmayr/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:98
  #9  0x4202147b:0x3fcbbed0 in Audio::setDefaults() at .pio/libdeps/esp32s3/ESP32-audioI2S/src/Audio.cpp:315
  #10 0x420218f3:0x3fcbbef0 in Audio::connecttohost(char const*, char const*, char const*) at .pio/libdeps/esp32s3/ESP32-audioI2S/src/Audio.cpp:538 (discriminator 2)  
  #11 0x42003599:0x3fcbbf60 in audioTask(void*) at src/audiotask.cpp:64

[... cut ...]
19:26:27 action: ...  current volume is 8
19:26:27 RESET_REASON:Reset reason can not be determined
19:26:27 WiFI_info:   RSSI is -46 dB
19:26:27 ftpServer:   Ftp server waiting for connection on port 21
19:26:28 webSrv: ...  File not found "/logo/.jpg"
19:26:29 BT-Emitter:  KCX_BT_Emitter not found 
19:26:33 DLNA_server: [0] 192.168.178.24:50001  DS213J_AL
19:26:33 DLNA_server: [1] 192.168.178.29:2869  WIN10ALF: Hoeckmayr:
19:26:33 DLNA_server: 2 media server found

Hilft das zur Fehlersuche weiter? 
Danke & Gruß, 
Alfons

TubaFonse, 2024-04-04:
Problem gelöst, erledigt mit #389
Danke & Grüße, Alfons

closed 2024-04-04 by TubaFonse