schreibfaul1/ESP32-MiniWebRadio #351: Wroom and or vs1053 No longer suitable for latest builds. What is a suitable HW Baseline. Issues 319, 222

RobinMosedale, 2024-02-18:
Dear Wolle!,

I'm delighted to have discovered your work. Like many, I'd become frustrated with the BBC moving away from ICY to HLS playlists.
I'd experimented with other ESP32 web radios, all worked well, but unable to handle the dynamic playlists used by the BBC. My commercial Web Radios also required subscription or indeed hadn't caught up with te BBC switchover.

So I was delighted that you'd integrated this in your suite.

However, I seem to have chosen the most problematic HW baseline: ESP32 WROOM/VS1053.
The ILI9341 seems solid with its SD (no resistors)

Like many I'd struggled with the SD Reader, but eventually I found that a 10k resistor on D0 pulled up to 3.3 solved it on my un-resistored variant of the ILI9341.

Having overcome that, a further issue 319 was encountered where the VS1053 fails to initialise, and the display sits at the welcome screen. Various resistors on SCK to pull down were experimented, and found that a 10k SCK to ground allowed the VS1053 to initialise. Once, thereafter repeated failures. However, if it succeeded, then I encountered Issue 222, Miniwebradio doesn't work properly without PSRAM. The suggested removal of the quoted patch (from both ESp32 and ESP32S3 configurations have no effect.

So it seems that only older VS1053 V1.2 are available , and not suitable for build. Moreover, WROOM variants without SPRAm are similarly unsuitable.

Perhaps the config files should be amended, by either removing those options, or warning that only upto Build xxx will work with them
I wonder if you could please recommend the simplest, cleanest configuration. 
I'm happy with my ILI9341(no-resistor SD).

I think that I can get my hands on a Dollatech T8 v1.7 esp32 WROVER which is an S3 with 4mb psram.
Ditto PCM5102 I2s fairly easily.
Would these with the ILI9341 form a simple clean build?

Eventually I'll need to decouple whatever output for Line Input to my HIFI Amp, or build a suitable Class D on the back end.

I'd be most welcome for your advice, but moreover, thank you for your efforts

Robin

schreibfaul1, 2024-02-19:
Hello Robin,
thank you for your report. You have overcome quite a few difficulties. I'm amazed that you and other users need pull-up resistors to operate the SD_MMC. I have often connected the SD card directly, just a wire connection without any other components. The VS1053 often causes problems, depending on which revision of the chip is installed on the board. I therefore recommend using an external DAC. This is easier to use and offers more possibilities. ESP32 with PSRAM, ILI9341 and PCM5102 will work properly.

steve6375, 2024-02-19:
#define DECODER             0

Why is this set to 0 when VS1053 is not supported !

closed 2024-04-04 by github-actions[bot]