schreibfaul1/ESP32-MiniWebRadio #353: Problem with ESP32 WROOM 32D and IR remote

tzvetan65, 2024-02-19:
Hello everyone,
I have a problem with IR remote control, works uncertain or even randomly. I changed a few transmitters and receivers, but the problem remains. Perhaps the reason is that I use ESP32 Wroom 32D and VS1053 and as the author writes memory is insufficient. Is it possible to sacrifice beauty because of a functionalist by removing VU Meter and how it can be done. Would this help with more free memory? And maybe there is another option. I will be grateful for ideas for having a full remote.

tzvetan65, 2024-02-19:
Hello again,
  Maybe I'm wrong for insufficient memory. I have now found that as I divert the transmitter away from the receiver, the quality of the connection is significantly improved - it works almost 100%. Before that, I tried to direct the transmitter as much as possible to the receiver, which misled me. Obviously, when directly targeted, the signal is very powerful and must be restricted or filtered. I soldered a pull-up resistor 10k between +5 and GPIO35, but there is no improvement.
   Has anyone had a similar problem? Any ideas???

schreibfaul1, 2024-02-19:
Yes, that depends very much on the design. The IR receivers usually have a metal encapsulation, but not all of them. It is important that the distance to the WiFi antenna is as large as possible. If this is not possible, you can try to shield the IR receiver from the WiFi antenna. Antistatic foil (the grey semi-transparent bag), as used for packaging electronic components, is best suited.

tzvetan65, 2024-02-19:
Hello, thanks for the quick answer and especially for the perfect development. Everything is easy and comfortable to work with and most importantly, you are constantly improving it.
   My receiver is SHARP GP1U721Q metally encapsulated, I also tried with VS1838B and TSOP1333, but SHARP it works best. I will look for connection schemes for different receivers and experiment.
   Greetings from Bulgaria.

closed 2024-04-04 by github-actions[bot]