schreibfaul1/ESP32-MiniWebRadio #354: Not a big problem. but it anoyed me. RTIME trying 10 times and reboot fix using configTzTime()

Linkenelis, 2024-02-24:
using configTzTime(RTIME_TZ.c_str(), NTP_Pool_1, NTP_Pool_2, NTP_Pool_3); 

and putting this in platformio.ini under build_flags=

        -DNTP_Pool_1='"europe.pool.ntp.org"' ;note the double quotes
        -DNTP_Pool_2='"pool.ntp.org"' 
        -DNTP_Pool_3='"time-a-g.nist.gov"' 
(you could use #define NTP_Pool_X in common.h as well)

and comment out a few lines that are done by the configTzTime function like
-sntp_setoperatingmode
-char sbuf[20]
-sntp_setservername
-sntp_init
-setenv
-tzset

Now it tries 3 servers, takes a bit longer per count as it tries 3 servers. But it rarely goes to 2/10 (so far).

boolean RTIME::begin(String TimeZone){
    if(TimeZone.length() == 0) return false;
    RTIME_TZ=TimeZone;
    if (RTIME_info) RTIME_info("Initializing SNTP");
    configTzTime(RTIME_TZ.c_str(), NTP_Pool_1, NTP_Pool_2, NTP_Pool_3);
    //sntp_setoperatingmode(SNTP_OPMODE_POLL);
    //char sbuf[20]="pool.ntp.org";
    //sntp_setservername(0, sbuf);
    //sntp_init();
    return obtain_time();
}

void RTIME::stop(){
    sntp_stop();
}

boolean RTIME::obtain_time(){
    time_t now = 0;
    int32_t retry = 0;
    const int32_t retry_count = 10;
    while(timeinfo.tm_year < (2016 - 1900) && ++retry < retry_count) {
        sprintf(sbuf, "Waiting for system time to be set... (%d/%d)", retry, retry_count);
        if (RTIME_info) RTIME_info(sbuf);
        vTaskDelay(uint16_t(2000 / portTICK_PERIOD_MS));
        time(&now);
        localtime_r(&now, &timeinfo);
    }
    //setenv("TZ","CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00", 1); // automatic daylight saving time
    //setenv("TZ", RTIME_TZ.c_str(), 1);
    //tzset();
    localtime_r(&now, &timeinfo);
    strftime(strftime_buf, sizeof(strftime_buf), "%c", &timeinfo);
    if (RTIME_info) RTIME_info(strftime_buf);

    //log_i( "The current date/time in Berlin is: %s", strftime_buf);
    if(retry < retry_count) return true;
    else return false;
}

schreibfaul1, 2024-02-25:
Thanks for your suggestion, I have integrated this in the V3.0.Branch

Linkenelis, 2024-02-25:
Nice! Thanks.

Don't forget to put this in platformio.ini under build_flags=

    -DNTP_Pool_1='"europe.pool.ntp.org"' ;note the double quotes
    -DNTP_Pool_2='"pool.ntp.org"' 
    -DNTP_Pool_3='"time-a-g.nist.gov"'

closed 2024-02-25 by Linkenelis

schreibfaul1, 2024-02-25:
Thank you, I have done that, I have a [common] section for ESP32 and ESD32-S3

https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/Arduino-V3.0/platformio.ini

Linkenelis, 2024-02-25:
You are way ahead of 💯