schreibfaul1/ESP32-MiniWebRadio #356: Nice new font in V3!

Linkenelis, 2024-03-02:
I see you adjusted the range and lookup.
Note that between Greek and Cyrillic you have 0 to 48 again and they should be 0. No problem if never used.
If you want me to do the Arial again (or something else) just shout. Happy to help.

schreibfaul1, 2024-03-02:
Thanks for your help, that must have happened when copying, I'll correct it soon [image]

I activated Arduino V3.0.0/IDF V5.1.2 today as a test. Currently only works with the precompiled Arduino without menuconfig. WiFi seems to be faster, I can now copy a file more than twice as fast. I don't know if there are any subsequent errors, I tested it with ESP32 and S3.

Linkenelis, 2024-03-03:
OK. I have a busy week ahead, but after that I will redo all the Font files following the correct lookup. and using the original ranges:
 * Range:  0x20-0x52F
 * 0x20 -0x7E  Basic Latin
 * 0x0A -0xFF  Latin Supplement
 * 0x100-0x17F Latin Extended A
 * 0x38E-0x3CE Greek   minus the 0x3A2, 0x38E-0x3A1 + 0x3A3-0x3CE
 * 0x400-0x45D Cyrillic

Right?

On yesterdays version, with the new font I found no problems.
I have on my own version OTA (I saw it in the new partitions files) implemented (ffat is not used anyway), so it can be updated without usb connection. I tested 16MB (smaller ffat), but should also work for 8MB (but without ffat)
[miniwebradio8MB_OTA.csv](https://github.com/schreibfaul1/ESP32-MiniWebRadio/files/14472979/miniwebradio8MB_OTA.csv)
[miniwebradio16MB_OTA.csv](https://github.com/schreibfaul1/ESP32-MiniWebRadio/files/14472980/miniwebradio16MB_OTA.csv)

in common.h add: #include <ArduinoOTA.h>     //Linke Nelis: Add the OTA library
in main.cpp -> setup() add: ArduinoOTA.begin();     //Linke Nelis: Start OTA
in main.cpp -> loop() add: ArduinoOTA.handle();    //Linke Nelis: Check if OTA needs handling
in platform.ini _> [env] add :
upload_protocol = espota                ;update code over WiFi
upload_port = 192.168.2.33             ; ip of miniwebradio
upload_flags = --host_port=55910

make sure in platformio to set Serial port to **auto**
[image]

schreibfaul1, 2024-03-03:
Hi @Linkenelis,
this seems to be the case with the character sets.
I had to adapt the partition tables as the old ones do not work with the new IDF.
Apparently the otadata partition with 2KB is absolutely necessary even if OTA is not used.

I wanted to use FFat instead of the SD_MMC, all files, logos and images could be stored in a /data folder and would be loaded automatically by the IDE. Then nothing would need to be unpacked to the SD card. Unfortunately this doesn't work with 4MB Flash, so I have discarded it.
I have never done anything with OTA, I will have a look at that soon. Let's see how it works, maybe that's a good idea.

Linkenelis, 2024-03-03:
@schreibfaul1 ,
Yes both Garamond and Times New Roman. FreeSerifeItalic had them all and more. Fonts directory in Windows 11 has  quite a bunch of fonts. Checking Garamond it also misses 0x400, but Times New Roman and Arial have it...
Checking all others, it is random as far as I can see. Clearly Times and Arial are among the most universal, with regular, italic and bold versions. So I have to look in to this further, for the other fonts. Garamond comes in many flavours, but I have yet to find the right one with the right glyphs.

FFat is less than12 MB on a 16 MB esp32 and less than 4 MB for an 8 MB esp32
You would be hard pushed to get it all in there with say 1000 jpg  * 3 kB + 500 kb .png, so 8 MB version will not do.
Only 16 MB version and not using "m" and  "s" and a bit of trimming here and there,  ffat has a chance. But what is the advantage? Only saving an SD card if you are not using audiofiles?

schreibfaul1, 2024-03-04:
You're doing a lot of work with the fonts. If a character is not available, it will certainly rarely be used. I don't know whether Ѝ ѝ is commonly used.
In the meantime, I have implemented the OTA suggestion. There is now a new selection option.
[image]
It seems to work for me, who knows if it always does. And of course, flash must be >= 8MB

commit 59f87c09 (2024-03-02): fix #356
closed 2024-03-05 by schreibfaul1