schreibfaul1/ESP32-MiniWebRadio #357: What I burned - please for advice

tzvetan65, 2024-03-03:
Hello,
Due to the noise at the amplifier output, I tried to separate the power supply into the ESP32 Wroom 32D and VS1053B. Apparently the two power supplies were not galvanically untied because the sound was lost and the ESP32 was restarted. The screen loads the graph, connects to the Internet, but not UV Metera has no movement. After the restart, the consumption of ESP32 was about 600ms, the display charges, but when it is used to manage it sometimes restarts again. I turned off the VS1053B, the consumption is about 400mm, it is restarted again, but rarely.
Can I check somehow what burned? If you need new modules what you would recommend, I see that it is recommended for ESP32 to be with PSRAM and the VS1053B is not recommended for use.

schreibfaul1, 2024-03-03:
That doesn't sound good, 400mA for the VS1053B is too much. Although, if your second voltage source does not exceed 5V and you rule out polarity reversal, little can actually go wrong. In principle, everything digital can be supplied from one voltage source without any problems, i.e. VS1053B + display + ESP32. The situation is different with a connected amplifier. AGND and GND are not the same thing. Then, in the simplest case, an additional power supply for the amplifier is advisable.
If the VS1053B is defective, it is better to use an external DAC such as the PCM5102A. Or 2x MAX98357A for loudspeaker operation.
Have a look at the V3.0 branch below the master in this repository. You can connect a KCX BT emitter there and connect to a BT headphones, BT speakers or assistants such as ALEXA.

tzvetan65, 2024-03-04:
Hello,
   Thanks for the answer, but I didn't write well the question I might have misled. In fact, one power supply is on ESP32, VS1053 and the display and the other is for the PAM8403 amplifier. Consumption 600m is common to ESP32, VS1053 and display, and 400mm is common to ESP32 and display. The output of the VS1053B and the input of the amplifier are separated by a capacitor of 47 microfarads, but still something has drilled. I asked for another version of ESP32 because sometimes it "forgets" to send to the display the graphics of the sound control buttons and the choice of a station. You have to click in the empty place where the button should be and then it appears, but only the one you click on, and so each next. I think this is because of the little memory of ESP32 and if so I would even change it to be healthy.
   I will be grateful for your recommendations.

clear-sky, 2024-03-06:
Hi -in vs1053 modules if audio ground connect to system ground it will resume hi current and main chip goes very very hot 
after several minutes high temperature will damage the module!

closed 2024-04-20 by github-actions[bot]