schreibfaul1/ESP32-MiniWebRadio #359: Feature request: Non-touch version + buttons

FedericoBusero, 2024-03-05:
I would like to adapt the software to a esp32-s3 hardware platform. It has
- 296x240 pixels. Is this possible?
- it doesn´t have a touch screen, but up/down/left/right is possible, next to 6 function keys

Did someone already try to add support for buttons on a non-touch display? Other suggestions?

clear-sky, 2024-03-06:
Hi 
a display with 296 x 240 pixels is very small possible but does not worth trying
i had asked several times from author to enable project fully with ir but seems he likes touch screens much more !!!
due to #245 and #309

Arne65, 2024-03-06:
Hello another user had a similar request.
There was the answer from Wolle.

void loop(){
...
    static bool bt = false;
    static uint32_t ms = 0;
    if(digitalRead(35) == LOW && bt == false){
        bt = true;
        nextStation();
        ms = millis() + 200; // wait 200ms
    }
   if(bt == true && ms < millis()) bt = false;
....
}
````

Maybe this will help you.

The rest you would have to adapt.

closed 2024-04-20 by github-actions[bot]