schreibfaul1/ESP32-MiniWebRadio #361: some 320k stations does not play correctly in V3

clear-sky, 2024-03-07:
Hi dear wolle!
in v3 some 320k stations are corrupted! for example:
http://us3.internet-radio.com:8342/stream

schreibfaul1, 2024-03-11:
Hi clear-sky, you will have noticed that the "menuconfig" entry in VS Code is disabled. We are currently working with the precompiled Arduino version. However, for some radio stations it is necessary to make changes in the WiFi configuration. But unfortunately there is still (at least one) bug that prevents this. But this will work again later.

clear-sky, 2024-03-11:
Hi dear Wolle! 
thanks for the answer
i am linux user as you ! 
but do not trust microsoft (softwares) at all - i have left windows and all related shits forever!
i use pio via command line thus did not know that

clear-sky, 2024-04-25:
Hi dear Wolle!
there is a serious problem in another station !!! it has AAC format at 128K --please test it
http://hts06.kshost.com.br:10182/;

schreibfaul1, 2024-04-25:
This is an AAC stream with "Spectral Band Replication". SBR requires additional memory, which can only be allocated to PSRAM due to its size. Unfortunately "menuconfig" does not work with V3/IDF5.1 at the moment. A temporary solution would be to deactivate SBR, in aac_decoder.h line 11

clear-sky, 2024-05-07:
Hi dear wolle!
please test this stream:
https://stream.zenolive.com/gxdq0awu30duv.aac

closed 2024-06-22 by github-actions[bot]